flandreas/antares #390: Crash during Save,  and during  import of exported schematic

PecanacKosta, 2022-06-01:
Dear sirs

After modifying the circuit using cut,  delete and  paste, and re-wiring some lines, it was not possible to save the schematic. I exported the schematic, but when I try to import it, an error occurs that the circuit cannot be opened.

Before the modification, everything worked fine.

Exported archive is in attachment.

Regards, 
Kosta

[Project.zip](https://github.com/flandreas/antares/files/8810234/Project.zip)

flandreas, 2022-06-01:
@KostaCoffee90 Thank you for reporting this.

By "it was not possible to save the schematic", do you mean that the save action wasn't enabled, or do you mean that an error occurred while saving? From the issue's subject, I'm assuming that an error occurred during save. Did you copy the error description from the error dialog to the clipboard or export the log file? Each of this would help to investigate the cause of the problem.

Either way, I will analyze the attached project archive and see if I can find any traces there.

Are you currently stuck with your project? Note that you can also use the function "Show History" in the popup menu of a circuit in the explorer tree to restore any older version of circuit.

KostaCoffee90, 2022-06-01:
Thanks for taking the time to solve my problem. It would be nice if a similar problem would be more clearly diagnosed within the schematic editor

Best regards

KostaCoffee90, 2022-06-01:
> Obviously my description of the problem was not clear enough ... But you
> have already solved it for me, thanks.
>
> It would be nice to fix a bug in the software - it's not a point that you
> patch the project archive to every user with a similar problem.

By the way

D-FF with PRE and CLR ( active HIGH or active LOW , both types) is not edge
sensitive, as it should be.
Unlike the ordinary D-FF, a  pulse generator is missing in the internal
realization.

For example, for that reason, shift registers with  asynchronous
parallel-in  cannot be realized "realistically" - it is necessary to add an
external pulse generator in order to prevent race condition
This is not a big problem, but it brings a certain amount of insecurity to
the students. (When Antares is used as a teaching tool)

Best Regards

flandreas, 2022-06-02:
@KostaCoffee90 Regarding the "crash while saving" problem: Sure, I'll have to fix the bug. That's why the log file would have been useful. Fixing the corrupt circuit archive was just an additional customer service to help you.

flandreas, 2022-06-02:
@KostaCoffee90 Regarding the missing edge-triggering of the "D Flip-Flops PRE/CRL": Thank you for reporting this. I will extract that one to a separate issue so I can handle it individually.

flandreas, 2022-06-02:
@KostaCoffee90 I found that the circuit corruption is related with the wire between input D of the "D Flip-Flop" at the bottom-right edge of circuit "New Circuit" and the AND gate (wire ID 252). There is another (invisible) wire that is also connected to the same D input, which is illegal and the source of the problem.

Do you remember how this particular wire came into existence? You've mentioned copy/paste and manual rewiring, and the bug is likely to be found in the corresponding code. I would greatly help to know in what order you've executed which action related with this particular wire. I know that's a lot to ask, but maybe I am lucky and you can remember it.

KostaCoffee90, 2022-06-03:
Dear Andreas

I am very willing to help as much as I can, and feel free to ask me about
any kind of assistance. I intend to suggest a few more things for you to
consider.

I don't have enough time at the moment, I'm hired as a court expert in a
civil dispute, and that turns out to be worse than being accused!

Since I have over 30 years of design experience (I started with the P-CAD
tool back in 1990), I got used to bugs and figuring out how to overcome
them, so in this case I didn't pay too much attention to the sequence of
steps that led to error.

I cut parts of the circuit and added them to the others, that's all I
remember

so at the moment I can't be of any help in this case.

But as soon as I have more time I will try to do similar activities and
cause a mistake, and then I will try to make a detailed report.

best regards

flandreas, 2022-06-06:
@KostaCoffee90 Thank you, your support and participation is much appreciated!

flandreas, 2024-05-19:
Silent issue, close.

closed 2024-05-19 by flandreas