flandreas/antares #401: Auto generating Symbols incompatible with already created Symbols

leonuetzel, 2022-06-19:
Upgrading to a new Version with auto-generating Symbols leads to Crashes when working with Symbols that have been created in previous Versions. Maybe this is related to the Blue Circle with a Cross that doesnt exist anymore in Antares Versions > 27

flandreas, 2022-06-20:
@leonuetzel Thank you for reporting. Do you have an error stack trace or a log file at hand?

flandreas, 2022-06-20:
@leonuetzel I did the following tests:

- Install 0.27.1
- Create circuit A with manually edited symbol, save
- Create circuit B and add subcircuit A, save
- Install 0.29.1
- Open B => Everything looks fine
- Create circuit C and add A as subcircuit, save => everything looks fine
- Connect A in C with new input and output component => Symbol of C is correctly adjusted
- Open A and go to symbol editor: Blue origin indicator is correctly displayed

So, I'm not able to reproduce the issue you're describing. In all mentioned circuits, I see the blue origin indicator when using 0.29.1.

May I ask you to help me to reproduce the error? Take a minute, install 0.29.1, and do again what you meant by "when working with Symbols that have been created in previous version". Don't save anything so that you can safely revert to 0.27.1 afterwards if necessary.

When the error occurs, please copy the error description to the clipboard and paste it here.

leonuetzel, 2022-06-20:
I will try that, but right now i have to finish the Project im working on in Antares (got a Deadline).
In 3 Weeks i will have Time to reproduce.

flandreas, 2022-06-20:
Okay, good look with your deadline!

flandreas, 2022-06-26:
@leonuetzel I was able to reproduce the problem.

Some problems occur if parts of the symbol are manually deleted, e.g. the automatically generated symbol block in symbol style "Wide". Under some circumstances, this deletes the blue origin indicator instead of the selected parts.

The cause of the problem is that some components (origin indicator, label) didn't save an ID in the circuit file. Such components did all receive ID 0 when read from the file. Since deletion is based on ID's, the first component with matching ID is deleted. In case of 0 IDs, this could be any other component with ID 0.

We fixed the code for the Components that didn't store their ID, and we implemented self-healing logic that applies new IDs to Components without an ID when reading them from files.

Please tell me what your experiences are with these fixes after you've upgraded to the release that contains theses fixed (probably the upcoming 0.29.2).

closed 2022-06-27 by flandreas