flandreas/antares #404: Propagation Delay of NAND is incorrectly simulated

leonuetzel, 2022-06-20:
[image]
The NAND in this Subcircuit shows a Delay of 5ns on the second incoming Signal Wave.
Reproduce with Starting Simulation, set "Pulse" to 3, then press "EN" and wait for the 2nd Wave to reach the Subcircuit. The NAND propagates the Signal after ca. 5ns, instead of the set Value of 20ns. Its clearly observable by watching the 2nd Wave propagating through the NOT-Delays, where the 2nd Wave is shorter than the 1st (which took about 10 NOT-Delays instead of ca. 2-3 for the 2nd Wave)
Folder "Parts->Clock" -> Circuit "Clock"
[FPU.zip](https://github.com/flandreas/antares/files/8943994/FPU.zip)
.

flandreas, 2022-06-23:
@leonuetzel Thank you for reporting this. I'm analyzing...

leonuetzel, 2022-06-23:
Ok sry then for bothering you, should have looked more carefully before opening an Issue.

closed 2022-06-23 by leonuetzel

flandreas, 2022-06-23:
@leonuetzel No problem, I'm always glad when I can help.