flandreas/antares #417: Vertical sidebar tabs sometimes become horizontal

aquamiau, 2022-07-06:
I have not been able to find out the cause for this yet, but once it happens screen estate is lost and there doesn't seem to be a way to reset it other than restarting the app.

[image]

flandreas, 2022-07-06:
@aquamiau Yes, thank you for reporting this. I've experienced this problem a long time ago myself, but never again since then.

Can you guess what was causing this problem in your case? Any user interaction prior to it? Or did it occur right after starting the application?

aquamiau, 2022-07-09:
Disconnecting or connecting a secondary screen seems to be one of the triggers, but I've also seen it happen without that.

flandreas, 2022-07-09:
@aquamiau Tried to fix it by using the official style of implementing UIDelegates in Java. Before, it was kind of a hack.

I'm not sure though if this fixes it. Please report back if you see the problem occurring again after upgrading to 0.30.0.