flandreas/antares #422: "Wired OR" to make book "But How Do It Know" possible?

flandreas, 2022-07-10:
I was recently trying to follow along the book "But How Do It Know" by J. Clark Scott using Antares to build the circuits described in the book. After all, when you write software, you tend do get stuck in software design issues and loose the view on aspects of challenges in your domain. It is therefore recommended to once in a while use your own software on a larger project just like your users do. In software engineering, this principle is denoted by "Eating your own dog food".

However, I've quickly noticed that I wouldn't get very far in the book, because the book doesn't contain the concept of "undefined / floating" signals. There are multiple components connected to the same bus, and components that shouldn't write to the bus can be disabled, but disabling components is implemented by simple AND gates, resulting in disabled components asserting 0 signals to a common bus. This of course leads to wire signal conflicts during simulation with Antares, when other components assert "real" signals containing 1.

It seems that the book implicitly relies on "Wired OR" behavior. Consider therefore to provide a feature in Antares allowing you to specify a "Wire behavior" per project. The default would be the standard conflict behavior as currently implemented, but you could also choose "Wired OR", which would simple OR all signals currently asserted to the same wire.

Kitick, 2022-08-08:
This reminds me of #155 that I posted a while ago, where I was trying to connect output wires together to essentially "OR" them on the wire as the name implies.

An issue being that if two outputs assert a state that can cause problems with the circuitry that isn't represented digitally. I agree having the option would be great, but would that cause problems if the circuit were to be built IRL? Correct me if I'm wrong, the book seams to focus on pure digital circuits, like you said their isn't a "floating value" because that's inherent to how a real circuit works. Not pure digital.

Also I totally agree, developers that use their own software will be able to improve it in so many ways greater than if not.

flandreas, 2022-08-08:
@Kitick You are right, #155 was about this very topic, but since I'm not an expert, I wasn't able to follow your discussion with @richardabendroth in all details. It was not until I've tried to implement the circuits in the book I started to wonder why they are even supposed to work without three-state-circuits.

Some googling showed me that the magic word was "open collector" (or "OpenDrain" in case of FETs) already brought up by Richard without me noticing then. So, the design of this feature #422 is based on the assumption that components can provide "Open Drain" outputs that are internally connected to a pull resistor, and you are allowed to connect multiple such outputs together to achieve "Wired-OR" logic.

See https://en.wikipedia.org/wiki/Open_collector on the English Wikipedia. Interestingly, the German Wikipedia has an dedicated article "Wired-OR" which explains the concept IMHO a little bit more understandable, at least for my application.

BTW, nice to see you around here again!

Kitick, 2022-08-12:
This is interesting, so basically chips are designed with a way to have that pull up resistor to allow them to connect to a main net. I feel the biggest issue would be backflow, so this would only be appropriate for a bus setting, and would need to be carful not to allow one output to flow back into an input, as Richard mentioned.

One other disadvantage that Wikipedia said was an increase in power consumption, since the resister is just dissipating energy. But that's not simulated with Antares. when I took my first circuits class in college I was reminded of this and my professor went with using tri states, because his logic was if you have things connected in a bus without them, a 1 would be forwarded, and get shorted to ground though the output of another gate. With the pull up resister, it still does that, just gets dissipated though the really strong resister. Making most of the energy flows though the circuit, not back to ground. Hence the increase in power consumption, since some energy is just getting dissipated. This also increases the heat output.

I think this is a difference in design philosophy, because it seams both work and are used, its more of, do you want more expensive chips, but more power efficient, or cheaper chips/more optimized wiring, but increased power consumption and heat output. At least that's how I see it currently, I could be missing things.

I recently got back into circuits and I thought to check to see how the development was going, awesome work so far! Its amazing to see how this program has grown. Its UI and interface is really nice to work with compared to Logicism, and how building and debugging circuits is super easy. Really the biggest fault is not having that developers license, but I totally understand the cost and time involved with it. Have you thought about having a way for people to donate to help with development?

flandreas, 2022-08-12:
@Kitick I agree with you on the "Wired OR" issue. Both design have their application, but "Wired OR" has higher power consumption and slower signal switching times. I think the author implicitly used "Wired OR" to avoid having to explain three-state-logic.

Thanks for your feedback. Regarding the developer license (app signing), I'm trying to have the apps signed by release 1.0.0. I've already done it for macOS (which was quite a pain), and I'm also trying to do it for Windows.

Well, regarding donations, thanks your kind question, very much appreciated. I will think about it.

closed 2022-09-09 by flandreas