flandreas/antares #498: Arbitrary Bitwidth for IO (and params)

daniel-endraws, 2022-10-28:
I think it would be nice if I could input a number in. Currently if I put `11` in the expression tab, it doesn't show the bitwidth in the editor. Thanks!

flandreas, 2022-10-28:
@daniel-endraws Thank you for your input.

Antares currently supports only certain predefined bit widths, i.e. the one suggested in the dropdown menu. 11 is not among them. The reason for this is that it was easier this way to implement IO views based on nibbles and the "Fan out" property of splitters/combiners.

If you enter the constant 11 as a bit width expression, the expression is internally evaluated and rejected, because it doesn't resolve to a supported bit width. It is a bug that Antares doesn't say that this is a unsupported bit width. I will open a bug issue for this.

I understand the requirement, and I think it makes sense to support any number between 1 and 64. Accepted as feature request.

In the meantime, as a workaround, you could use the next larger available bit width and ignore the top bits.

daniel-endraws, 2022-10-28:
Thanks for the quick response! It would also be nice if the fan out could be arbitrary divisions specified by some sort of params.

daniel-endraws, 2022-10-28:
Sorry I think I accidentally closed the issue...

Thanks for the quick response!

flandreas, 2022-10-29:
Never mind

closed 2022-12-01 by flandreas