flandreas/antares #510: VHDL and verilog support

wmurbina, 2022-11-28:
Greetings.
It would be interesting to be able to create components using hardware description language like VHDL and Verilog, you could take as a basis the script module that Antares has.
It would also be interesting to be able to export a design to VHDL or verilog and test the operation on an FPGA.

Thanks.

flandreas, 2022-11-28:
@wmurbina Thank you for your request.
Regarding your first point, this could be done by implementing a parser for VHDL and interpret the resulting AST with the already available interpreter. Not so quickly doable, but definitely interesting, although to be honest currently rather of low priority.
Your second point has already been requested by other users, and I'm accepting it as a feature request. Would you rather have VHDL or Verilog first?

wmurbina, 2022-11-29:
Greetings.
Thank you for your replies.
I would be interested in working on VHDL first.

flandreas, 2023-08-18:
Started implementing VHDL export.

closed 2023-11-24 by flandreas