flandreas/antares #522: Importing libraries is not working

Hougaarden, 2023-02-04:
Pressing the "Import" button generates al foult:
[image]

Version: 1.2.0
Loading settings from 'C:\Users\Wilfried\AppData\Roaming\Antares\Antares.ini'
Loading preferences from 'C:\Users\Wilfried\AppData\Roaming\Antares\Antares.pref'
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'Subcircuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
java.lang.IllegalArgumentException: Comparison method violates its general contract!
	at java.base/java.util.ComparableTimSort.mergeHi(Unknown Source)
	at java.base/java.util.ComparableTimSort.mergeAt(Unknown Source)
	at java.base/java.util.ComparableTimSort.mergeForceCollapse(Unknown Source)
	at java.base/java.util.ComparableTimSort.sort(Unknown Source)
	at java.base/java.util.Arrays.sort(Unknown Source)
	at java.desktop/sun.awt.shell.Win32ShellFolderManager2.get(Unknown Source)
	at java.desktop/sun.awt.shell.ShellFolder.get(Unknown Source)
	at java.desktop/javax.swing.plaf.metal.MetalFileChooserUI$DirectoryComboBoxModel.addItem(Unknown Source)
	at java.desktop/javax.swing.plaf.metal.MetalFileChooserUI.doDirectoryChanged(Unknown Source)
	at java.desktop/javax.swing.plaf.metal.MetalFileChooserUI$5.propertyChange(Unknown Source)
	at java.desktop/java.beans.PropertyChangeSupport.fire(Unknown Source)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)
	at java.desktop/java.awt.Component.firePropertyChange(Unknown Source)
	at java.desktop/javax.swing.JFileChooser.setCurrentDirectory(Unknown Source)
	at java.desktop/javax.swing.JFileChooser.<init>(Unknown Source)
	at java.desktop/javax.swing.JFileChooser.<init>(Unknown Source)
	at ch.scorpion.jabbah.graph.library.p$c.a(SourceFile:69)
	at ch.scorpion.jabbah.base.j.actionPerformed(SourceFile:74)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.WaitDispatchSupport$2.run(Unknown Source)
	at java.desktop/java.awt.WaitDispatchSupport$4.run(Unknown Source)
	at java.desktop/java.awt.WaitDispatchSupport$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.desktop/java.awt.WaitDispatchSupport.enter(Unknown Source)
	at java.desktop/java.awt.Dialog.show(Unknown Source)
	at java.desktop/java.awt.Component.show(Unknown Source)
	at java.desktop/java.awt.Component.setVisible(Unknown Source)
	at java.desktop/java.awt.Window.setVisible(Unknown Source)
	at java.desktop/java.awt.Dialog.setVisible(Unknown Source)
	at ch.scorpion.jabbah.base.swing.c.d(SourceFile:105)
	at ch.scorpion.jabbah.graph.library.aO$b.a(SourceFile:53)
	at ch.scorpion.jabbah.graph.library.bX.a(SourceFile:27)
	at ch.scorpion.jabbah.base.j.actionPerformed(SourceFile:74)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

flandreas, 2023-02-05:
@Hougaarden Thank you for reporting this.

I cannot reproduce the problem. According to a google search, the problem seems to be Windows-specific and caused by an incompatibility of a search algorithms in Java's file chooser and the way Windows provides its files.

It seems like Windows "Quick access" feature duplicates files, which Java's file chooser can't work with.
I will look for a fix for this problem in the Java/Windows community.

In the meantime, you can try to clear your "Quick Access" in Windows. Some people who encountered this problem in other Java applications have reported that this could make the error go away.

flandreas, 2023-02-05:
Internal developer comment:
According to https://github.com/skylot/jadx/issues/1552, adding the Java option "-Djava.util.Arrays.useLegacyMergeSort=true" might fix the problem.

Hougaarden, 2023-02-05:
Dear Andreas,
Thanks for the quick reply.
I have cleared "Quick Access", but that did not work.
Maybe there is a way to add libraries manually?
In itself it is not such a big problem for me, I will wait and see what possible updates offer for a solution.

Regards, Wilfried Vermeire (Belgium)

flandreas, 2023-02-05:
@Hougaarden 
Dear Wilfried
You could try to do the following to add the library manually. Before trying this, make sure to create a backup of your data directory in C:\Users\Wilfried\AppData\Roaming\Antares.

- If you don't already have a custom library, create a dummy new one just to let Antares create the necessary file structures. Use File->Libraries->New to create a new library.
- Extract the library.zip file you want to import e.g. on your Windows desktop. This gives you a directory of the save name.
- Open that directory and find an inner directory with a name like "af4ddc6f...." (a long UUID). This is the library contents directory to be manually imported into Antares.
- Move that directory into the "libraries" folder of your Antares data directory, in your case C:\Users\Wilfried\AppData\Roaming\Antares\libraries
- Use a text editor and open the file "dictionary.xml" in C:\Users\Wilfried\AppData\Roaming\Antares\libraries to manually register your library.
- Copy everything between <libraryDirectoryEntry> and </libraryDirectoryEntry> (including these tags) and paste it below. This will create another directory entry between <entries> and </entries>
- In the copied <libraryDictionaryEntry>, replace the existing uuid='030e7e6e..." with the UUID from the name of your library directory to be added, such as uuid='af4ddc6f....'
- In the copied <libraryDictionaryEntry>, replace _id='1' or whatever number it has with the smallest number that don't yet exist. This number has to be unique. Im my example below, this is _id='2'
- In <name> and its <translation> element, replace text='My Library' or whatever the name of the existing library is, with the name of the library to be imported, e.g. "Manually copied library"
- Save the edited "dictionaries.xml"

Finally, restart Antares and try to "Open" the manually copied library.

Example dictionary.xml after copying `<libraryDirectoryEntry>`:

```
<?xml version='1.0' encoding='UTF-8'?>
<libraryDictionary _id='0'>
  <entries>
    <libraryDictionaryEntry _id='1' uuid='030e7e6e-f0b6-4d47-8110-34761f981f34' author='7e840175-f9c4-4886-a221-ef91e3493e27' visibility='private'>
      <name>
        <translation lang='en' text='Test'/>
      </name>
    </libraryDictionaryEntry>
    <libraryDictionaryEntry _id='2' uuid='af4ddc6f-3edb-434c-8884-0d3378376cb7' author='7e840175-f9c4-4886-a221-ef91e3493e27' visibility='private'>
      <name>
        <translation lang='en' text='Manually copied library'/>
      </name>
    </libraryDictionaryEntry>
  </entries>
</libraryDictionary>
```

Hougaarden, 2023-02-06:
Dear Andreas,

Thanks for the clear explanation. I've tried it with the SN74XX library and it works.
Thanks again. I look forward to further expansions and updates.

Kind regards, Wilfried Vermeire

Hougaarden, 2023-02-19:
Dear Andreas,

 I cleaned up my computer by deleting some unused applications.
Importing libraries and projects now works as it should.
So the "bug" was not in your program, but in my computer ;)

Kind regards, Wilfried Vermeire

closed 2023-02-19 by Hougaarden