flandreas/antares #545: CAD-style selection

damjes, 2023-03-27:
Currently (1.3.0), You can choose between two selection box scopes: completely and partially contained.

In CAD programs there is a rule, that selection box depends of direction. In Antares terms and CAD GUI it will be:
- left: partially contained and dashed green
- right: completely contained and solid blue

It would be nice to select "CAD style" or similar to "Selection box scope". Difference in looks could be nice, but it's not as important.

Here is some additional sources:
https://forums.autodesk.com/t5/autocad-forum/difference-between-left-cross-and-right-window-selection/td-p/2898188
https://www.youtube.com/watch?v=pqU7kJw1XMQ

damjes, 2023-03-29:
I am not sure, it seems nice, but also nonstandard.

flandreas, 2023-03-30:
Implemented the ALT variant.

closed 2023-05-28 by flandreas