flandreas/antares #557: Won't install in win11, win10 think its a virus

Ali-exe, 2023-05-14:


flandreas, 2023-05-16:
@Ali-exe Thank you for reporting this.

Regarding Windows 10: No. The blue info dialog you see is not a virus warning. It's an info Windows is displaying because Antares is using an intermediate certificate. You would see a "virus warning" if Antares wouldn't use a certificate at all. Omitting the warning would require an advanced certificate, which is very expensive, and which we can't afford yet.

Regarding Window 11: Okay, we weren't aware of that, thank you. We don't have Windows 11 at hand yet, but we will see what we can do to fix this.

LuminousXLB, 2023-05-31:
I'm using Win11, and I can install v1.3.0 and v1.4.0 smoothly. More information may be needed to reproduce the problem.

flandreas, 2023-06-01:
@LuminousXLB Thank you, I'm glad to hear that. We're on our way to build up a Windows 11 testing environment and we will proceed with this issue once the setup is completed.

richardabendroth, 2023-06-16:
I can confirm, that installing on Win11 will raise a warning that instructs the user to not execute the program. It can be circumvented the usual way. I think you miss a certificate here.

flandreas, 2023-06-17:
We have now a Win11 installation and can also confirm that  our certificate is not accepted by Win11 the same way it is by Win10. We probably need to check with the issuer of our certificate why this is the case.

flandreas, 2023-07-01:
After building release 1.5 on Win 11, the installation dialog looks the same as on Win 10.

richardabendroth, 2023-07-03:
Still the same on Win11 for 1.5.0

[image]
[image]

flandreas, 2023-07-06:
@richardabendroth Strange. I don't see this on my machine. Reopen.

flandreas, 2023-07-16:
@richardabendrot

 I've done some research. The problem is MS SmartScreen. Unless the app is signed with an EV certificate, which is expensive and only issued to registered companies, and even if an app is signed with a non-EV certificate, SmartScreen blocks the app until the app has built reputation of possibly thousands of downloads. Since app signing certificates have a validity of 1 to 2 years, and SmartScreen resets gained reputation to zero when the signature has changed, Antares will never be downloadable without SmartScreen warning.

On Windows, certificates simply don't make sense for non-commercial application, and I think I won't re-new my certificate after its validity has expired somewhen this August.

closed 2023-08-20 by flandreas