flandreas/antares #565: Feature Request: Enable dragging canvas by holding the space key

LuminousXLB, 2023-05-24:
Thank you for the fantastic tool.

I propose implementing a feature enhancement that enables users to conveniently drag the canvas by holding down the space key, allowing for seamless navigation and repositioning without the need for zooming in and out.

flandreas, 2023-05-24:
@LuminousXLB 

Are you working with a mouse or on a notebook with a trackpad?

When using the mouse, you can pan the view by pressing the middle mouse button (or the mouse wheel) while moving the mouse around.

When using a trackpad, you can pan the view by pressing ALT (or Option on macOS) while performing a scroll gesture on trackpad.

However, I've just tested this on a Windows notebook, and ALT + scroll gesture seems to not work. I will open a separate bug issue for this.

LuminousXLB, 2023-05-24:
Hi @flandreas

Thanks a lot for your information. That really helps.

LuminousXLB, 2023-06-01:
Hi @flandreas . I had some more experience with this now. 

I don't think pressing the mouse wheel would be a good experience. For some types of mouse, it needs much force to press that wheel. Using ALT with scroll would work but we can only scroll up/down, cannot left/right.

I've also tried using Opt+Trackpad on MacBook. It seems that when I scroll down, the canvas goes up.

I would still upvote for panning using Space+drag (or Ctrl+drag, to be consistent with Vivado block design).

https://docs.xilinx.com/r/en-US/ug893-vivado-ide/Using-Mouse-Strokes-to-Zoom-and-Pan

flandreas, 2023-06-01:
@LuminousXLB Okay, thanks for the feedback. I think I will introduce a user preference and include there your suggestions.

Regarding macOS trackpad, this will probably be influenced by the macOS "Natural scroll direction" setting on macOS. I will have look into that as well.

flandreas, 2023-06-30:
@LuminousXLB

Implemented a new user preference with the option to choose "ALT + press/drag left mouse button", in addition to the default "Press/drag middle mouse button".

Unfortunately, implementing SPACE key was not possible (at least not straight-forward), because the underlying Java system doesn't offer SPACE as modifier for MouseEvents.

closed 2023-07-01 by flandreas