flandreas/antares #576: Unexpected Error when Opening Oscilloscope Preferences

LuminousXLB, 2023-06-01:
[image]

```
Version: 1.4.0
Loading settings from '/Users/username/Library/Application Support/Antares/Antares.ini'
Loading preferences from '/Users/username/Library/Application Support/Antares/Antares.pref'
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
java.util.MissingResourceException: Missing translation 'antares.preference.SignalHistory.fill'
	at ch.scorpion.jabbah.base.R.a(SourceFile:76)
	at ch.scorpion.jabbah.base.R.a(SourceFile:29)
	at ch.scorpion.jabbah.base.k.a.d(SourceFile:90)
	at ch.scorpion.jabbah.base.k.c.a(SourceFile:130)
	at ch.scorpion.jabbah.base.k.y.<init>(SourceFile:1018)
	at ch.scorpion.jabbah.base.k.A.a(SourceFile:3138)
	at java.desktop/javax.swing.JTree.fireValueChanged(Unknown Source)
	at java.desktop/javax.swing.JTree$TreeSelectionRedirector.valueChanged(Unknown Source)
	at java.desktop/javax.swing.tree.DefaultTreeSelectionModel.fireValueChanged(Unknown Source)
	at java.desktop/javax.swing.tree.DefaultTreeSelectionModel.notifyPathChange(Unknown Source)
	at java.desktop/javax.swing.tree.DefaultTreeSelectionModel.setSelectionPaths(Unknown Source)
	at java.desktop/javax.swing.tree.DefaultTreeSelectionModel.setSelectionPath(Unknown Source)
	at java.desktop/javax.swing.JTree.setSelectionPath(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicTreeUI.selectPathForEvent(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicTreeUI$Handler.handleSelection(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicTreeUI$Handler.mousePressed(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicTreeUI$MouseHandler.mousePressed(Unknown Source)
	at com.formdev.flatlaf.ui.FlatTreeUI$1.mousePressed(SourceFile:247)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.WaitDispatchSupport$2.run(Unknown Source)
	at java.desktop/java.awt.WaitDispatchSupport$4.run(Unknown Source)
	at java.desktop/java.awt.WaitDispatchSupport$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.desktop/java.awt.WaitDispatchSupport.enter(Unknown Source)
	at java.desktop/java.awt.Dialog.show(Unknown Source)
	at java.desktop/java.awt.Component.show(Unknown Source)
	at java.desktop/java.awt.Component.setVisible(Unknown Source)
	at java.desktop/java.awt.Window.setVisible(Unknown Source)
	at java.desktop/java.awt.Dialog.setVisible(Unknown Source)
	at ch.scorpion.jabbah.base.swing.c.d(SourceFile:105)
	at ch.scorpion.jabbah.base.k.q$a.a(SourceFile:47)
	at ch.scorpion.jabbah.app.l.a(SourceFile:207)
	at java.desktop/com.apple.eawt._AppEventHandler$_PreferencesDispatcher.performUsing(Unknown Source)
	at java.desktop/com.apple.eawt._AppEventHandler$_PreferencesDispatcher.performUsing(Unknown Source)
	at java.desktop/com.apple.eawt._AppEventHandler$_AppEventDispatcher$1.run(Unknown Source)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
```

flandreas, 2023-06-01:
@LuminousXLB Thanks for reporting this!

Kitick, 2023-06-01:
I want to add I also get this error on windows. I thought it was because I didn't have an oscilloscope.

flandreas, 2023-06-02:
@Kitick No, it's really just a missing translation text.

closed 2023-07-01 by flandreas