flandreas/antares #582: Open-ended wire can sometimes not be reconnected

flandreas, 2023-06-08:
When deleting a connected component, the resulting open-ended wires can sometimes not be reconnected to another component.

Example:

[image]

1. Delete inverter
2. Reconnect the now open-ended wire at output of OR gate to pin of O2

=> The pin at O2 won't accept the wire

medennis34, 2023-06-10:
[image]
I think I have a similar problem.  I deleted a wire between the clock (set) and the buffer enable and there appears to now be an invisible short.

flandreas, 2023-06-30:
@medennis34 Sorry, I just saw your comment now.

Do you remember what the tooltip said when hovering over one of the wires in error state (the red ones)?.

flandreas, 2024-05-19:
Cannot be reproduced any more.

closed 2024-05-19 by flandreas