flandreas/antares #583: Random Error in "slow" simulation

richardabendroth, 2023-06-16:
Hi Andreas, 
you've put a lot of great work into Antares since the last time I've used it! I've encountered a bug when running a simulation. When running it in "slow" mode: 

[image]

Version: 1.4.0
Loading settings from 'C:\Users\**********\AppData\Roaming\Antares\Antares.ini'
Loading preferences from 'C:\Users\**********\AppData\Roaming\Antares\Antares.pref'
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=false
java.util.NoSuchElementException: distance
	at ch.scorpion.jabbah.a.t.c(SourceFile:83)
	at ch.scorpion.jabbah.a.t.b(SourceFile:57)
	at ch.scorpion.jabbah.graph.view.net.edge.y.a(SourceFile:1042)
	at ch.scorpion.jabbah.a.a.a(SourceFile:49)
	at ch.scorpion.jabbah.a.c.h(SourceFile:61)
	at ch.scorpion.jabbah.a.j.c(SourceFile:6121)
	at ch.scorpion.jabbah.a.k.invoke(SourceFile:1043)
	at ch.scorpion.jabbah.base.d.d.actionPerformed(SourceFile:14)
	at ch.scorpion.jabbah.base.o.c$a.actionPerformed(SourceFile:54)
	at java.desktop/javax.swing.Timer.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.Timer$DoPostEvent.run(Unknown Source)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

flandreas, 2023-06-18:
@richardabendroth
Richard, this is especially weird, because the drawing is clearly in edit mode, but there is still a bit flow animation drawn.

I guess you can't reproduce the bug with the current version of that circuit, right? If you can, I would love to have a copy of that circuit in order to analyse the wire structure at the output of that pulse generator.

If you can't reproduce it because you've changed the wiring in the meantime, may I ask you to create a ZIP file of the "history" directory in your working directory? I could then extract the version that produces the error and analyse it.

richardabendroth, 2023-06-19:
[Antares.zip](https://github.com/flandreas/antares/files/11786014/Antares.zip)

flandreas, 2023-06-22:
I can reproduce this problem by using version 16.6.23, 1:36:40 from the attached project, after manual workaround around bug #585 and #584 by deleting the wire to input port 1 of AND gate 512, then running the simulation with signal animation, clicking switch "STP" and waiting until the signal arrives at the output of pulse generator 459.

flandreas, 2023-06-23:
The problem occurs if the last segment of a wire leading to a junction is much smaller than any other wire of the same net, but only if the simulation speed is sufficiently fast. The simulation speed determines the sequencing step width when driving a bit along a wire, and if that step width is larger than a wire segment size, the observed exception can occur.

The bug was introduced when reworking signal flow animation on wires for the upcoming analog feature, where current flow animation on wire is needed (https://twitter.com/AntaresCircuit/status/1618526174560555008).

This is now fixed. Will be delivered with the next release.

closed 2023-07-06 by flandreas