flandreas/antares #584: "Dead" components

richardabendroth, 2023-06-16:
Sometimes, components become "dead": 

[image]

flandreas, 2023-06-17:
@richardabendroth Thank you for reporting this.

Do you have the feeling that it could be related with undo/redo? I guess that the problem persist, once it has occurred, when you save and reload the circuit. May I ask you to send me your project and the log file when the problem occurs the next time? This would be of great help to track down the root cause of the problem.

richardabendroth, 2023-06-17:
Hey Andreas,
I've not used undo/redo. It happened during connecting/reconnecting wires to the net behind it I think. I will certainly make a copy of the circuit when it happens the next time 😊

flandreas, 2023-06-21:
@richardabendroth I am able to reproduce an instance of this problem by playing with the project you've updated in #583. I can now analyse this.

[image]

richardabendroth, 2023-06-21:
Nice!

flandreas, 2023-06-22:
@richardabendroth I found out that the reason why the AND gate is dead is that the upper incoming wire (the one selected in the screenshot below) is connected with the AND on the view level, but _not_ connected on the model layer. Therefore, input port 1 of the AND gate is always Z, and the AND gate produces 0 instead of the expected 1.

[image]

I now need to find the operation in the code that produces this model inconsistency.

I know it's a lot to ask, but do you by any chance remember how this wire came into existence? I see from the project history that you've designed the entire block with CNT, DEC and the pulse generators at the left side of the circuit, and then moved it in and connected the parts. Was copy/paste involved? Undo/redo?

It's okay if you can't remember, I'm just trying to reduce the amount of areas I have to inspect while searching for the bug.

richardabendroth, 2023-06-22:
It's quite likely that I've used copy&paste. I tend to pick a component from the library when I need it the first time. For any additional component of the same type I tend to use copy&paste instead of picking it from the library because that is much faster. But I cannot remember if in this case the AND gate was a copy or picked from the library. But the probability of it originating from a copy&paste operation is quite high.

flandreas, 2023-07-09:
@richardabendroth I'm currently focusing on finding the cause of this severe bug.

- Identify possible cause hypothesis
- Implement diagnostics for the various hypothesis
- A new feature that let the user export the entire system state for uploading to a GitHub issue and later analysis

closed 2023-08-20 by flandreas