flandreas/antares #585: Error when deleting component

richardabendroth, 2023-06-16:
Version: 1.4.0
Setting DrawingView with 'New Circuit' to editable=false
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=false
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=false
Setting DrawingView with 'New Circuit' to editable=false
Setting DrawingView with 'Binary Counter 4 Bit LD/CLR' to editable=false
Setting DrawingView with 'Binary Counter 4 Bit LD/CLR' to editable=false
Setting DrawingView with 'Binary Counter 4 Bit LD/CLR' to editable=false
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=false
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=false
Setting DrawingView with 'New Circuit' to editable=true
Rollback transaction
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
Rollback transaction
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
java.lang.IllegalStateException: Net not connected with specified Port
	at ch.scorpion.jabbah.graph.model.c.d.b(SourceFile:73)
	at ch.scorpion.jabbah.graph.view.b.ek.c(SourceFile:93)
	at ch.scorpion.jabbah.graph.view.a.i.a(SourceFile:201)
	at ch.scorpion.jabbah.edit.c.d.a(SourceFile:176)
	at ch.scorpion.jabbah.graph.view.a.d.a(SourceFile:156)
	at ch.scorpion.jabbah.graph.view.a.d.a(SourceFile:94)
	at ch.scorpion.jabbah.edit.a.l$a.a(SourceFile:35)
	at ch.scorpion.jabbah.edit.a.j.a(SourceFile:41)
	at ch.scorpion.jabbah.base.j.actionPerformed(SourceFile:74)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Actions.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.SwingUtilities.notifyAction(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBinding(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processKeyBinding(Unknown Source)
	at java.desktop/javax.swing.KeyboardManager.fireBinding(Unknown Source)
	at java.desktop/javax.swing.KeyboardManager.fireKeyboardAction(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBindingsForAllComponents(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBindings(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

flandreas, 2023-06-17:
@richardabendroth Thank you for reporting this.

I see that the error occurred pretty quickly after you've opened the circuit. Do you remember what you did right before trying to delete the component?

I'm investigating around the suspicious "Rollback transaction" log entries. Possible action candidates are:
- Canceling creation of a new wire from an existing wire to a pin
- Reconnecting a wire from a pin to another pin, and releasing the mouse very near the origin pin

richardabendroth, 2023-06-17:
That's hard to tell from memory. I don't think it was the first option. But it might also have been a connection of a "free" wire end to a new wire or a component pin.

flandreas, 2023-06-22:
This is also caused by the model inconsistency in #584 "Dead Component": When deleting the AND gate, Antares tries to unconnect port 1 from the net, but the net complains that there is no connection.

flandreas, 2023-07-09:
Definitively the same cause as #584. I'm closing this as duplicate and investigate further in #584.

closed 2023-07-09 by flandreas