flandreas/antares #587: Error when starting simulation

richardabendroth, 2023-06-16:
Version: 1.4.0
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=false
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=false
Setting DrawingView with 'New Circuit' to editable=false
Setting DrawingView with 'New Circuit' to editable=false
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'New Circuit' to editable=true
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'Symbol' to editable=true
Setting DrawingView with 'SM54HC163 FlipFlop' to editable=true
Setting DrawingView with 'SM54HC163 FlipFlop' to editable=true
java.lang.NullPointerException
	at ch.scorpion.antares.model.l.q.<init>(SourceFile:43)
	at ch.scorpion.antares.model.l.q.<init>(SourceFile)
	at ch.scorpion.antares.model.l.q$a.a(SourceFile:35)
	at ch.scorpion.antares.model.l.B.a(SourceFile:33)
	at ch.scorpion.antares.model.port.DigitalPortImpl.getDefaultDigitalSignal(SourceFile:202)
	at ch.scorpion.antares.model.port.DigitalPortImpl.executionStarted(SourceFile:207)
	at ch.scorpion.jabbah.graph.model.f.d.b(SourceFile:164)
	at ch.scorpion.jabbah.graph.model.graph.GraphImpl.executionInitialize(SourceFile:223)
	at ch.scorpion.jabbah.graph.model.f.l.b(SourceFile:217)
	at ch.scorpion.jabbah.graph.model.graph.GraphImpl.executionInitialize(SourceFile:223)
	at ch.scorpion.jabbah.graph.view.G.a(SourceFile:1104)
	at ch.scorpion.jabbah.graph.view.I.invoke(SourceFile:1039)
	at ch.scorpion.jabbah.base.d.a.b(SourceFile:78)
	at ch.scorpion.jabbah.c.e.n.b(SourceFile:1462)
	at ch.scorpion.jabbah.graph.a.g.invoke(SourceFile:1092)
	at ch.scorpion.jabbah.base.N.a(SourceFile:25)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

richardabendroth, 2023-06-16:
Circuit causing the issue: 
[image]

flandreas, 2023-06-17:
@richardabendroth Thank you for reporting this. I can reproduce it.

flandreas, 2023-06-17:
@richardabendroth Found the cause and fixed it. Fix will be available with the next release.

closed 2023-07-01 by flandreas