flandreas/antares #590: Input/Output names when copying&pasting

richardabendroth, 2023-06-23:
When copying and pasting Inputs/Outputs, the newly assigned name is either I[n] or O[n]. While this makes sense when dragging a new Input/Output from the Library, it would be great when using copy&paste instead, when the name would be copied as well but maybe with a (2) or something appended. E.g. when copying an Input with name "Σ1" it would be great if the new name would be "Σ1 (2)" or even "Σ2" if somehow possible.

flandreas, 2023-06-24:
Thank you Richard, this is an interesting suggestion, I think we will implement this.

richardabendroth, 2023-06-24:
Another thought: You could instead expose the naming pattern as properties on the circuit for input, output, and input/output. Something like: 
Input Naming: I{n}
Output Naming: O{n}
Input/Output: IO{n}

74xx series IC's often have inputs like A, B, C, D and outputs with numbers, which would look like: 
Input Naming: {L}
Output Naming: {n,0} (0 being the starting number)

And one could also do things like "Σ{n,1}". 

And you should definitely add the "Input/Output" components to the Inputs and Outputs categories.

flandreas, 2023-06-30:
Requirement "And you should definitely add the "Input/Output" components to the Inputs and Outputs categories" extracted to separate issue #604.

flandreas, 2023-08-11:
Requirement "Expose naming pattern" extracted to separate issue #612.

flandreas, 2023-08-11:
Implemented first requirement:
- If name adheres to pattern "text,number", the pasted name will be "text,number+1", or the next available free number respectively
- Else, apply the pattern "Text (2)"

closed 2023-08-20 by flandreas