flandreas/antares #591: Editing properties of multiple identical components is limited to 8

richardabendroth, 2023-06-24:
When selecting more than 8 identical components, the "Properties" area becomes blank: 

[image]
[image]

flandreas, 2023-06-25:
@richardabendroth Yes, the number of components for which "Edit properties of multiple components" is possible is intentionally limited to 8. This feature is very computation intense, as it has to find the common denominator of the properties of all selected components. "Select All" in a circuit with hundreds of components would perhaps make your computer explode. I thought the number 8 was a nice IT number. Would you wish a larger number?

richardabendroth, 2023-06-25:
Oh I see. Maybe you should have a message in the properties area indicating this, like: "Bulk editing of properties is limited to 8 components for performance reasons" or "too many components selected".

flandreas, 2023-06-25:
You're right, I'll do that.

closed 2023-07-01 by flandreas