flandreas/antares #592: Allow subscript in component name

richardabendroth, 2023-06-24:
In component names, the !X and !(XXX) creates an overline for the respective text. However, it would be nice to have a similar syntax for subscript, e.g. "Q^A", resulting in Q<sub>A</sub>, "X^(!(DATA))", resulting in X<sub>D̅A̅T̅A̅</sub>.

flandreas, 2023-06-25:
@richardabendroth That's a great suggestion, I'll do that.

flandreas, 2023-07-01:
@richardabendroth I did some research on this one. Unfortunately, it turns out that this one is not as easy to implement as I was hoping.

The current implementation of overlines for negations is purely based on Unicode characters. The same approach seems not to work for subscripts, because Unicode (as far as I understand) does only support a couple of subscript characters: All numbers and some letters, those that are typically used in math expressions, such as i, e, x and so forth.

It turns out that Antares would need an entirely different approach for rich text that is more based on styling rather than Unicode only. For Java canvas, this could be AttributedText together with Graphics2D method. In Java UI, this could be HTML for rendering tree nodes etc. Browser canvas, which must also be supported in Antares, would need yet something different.

My current conclusion sadly is that this story is an epic.

flandreas, 2023-08-05:
@richardabendroth Implemented. I've also added superscript as well as bold and italic font style.

[image]

closed 2023-08-20 by flandreas