flandreas/antares #598: Mnemonics of XNOR gate are off

richardabendroth, 2023-06-28:
The wiring withing the mnemonics of the XNOR gate look off compared to the XOR gate: 

[image]

flandreas, 2023-06-29:
By "off" I guess you mean that the lower input of the XNOR is internally guided to the top of the gate.

Well, this is done intentionally. Let me explain.

1. The internal "double bar" switches are generally meant to be in rest with input 0, and pushed against an imaginary spring with input 1.
2. XNOR produces output 1 if and only if both inputs are the same.
3. Both internal "double bar" switches are at the same vertical position when at rest.
4. To achieve the XNOR behaviour, both "double bar" switches must move into the same direction when being pushed by a 1 input.
5. Therefore, the lower external gate input must be guided to the top of the gate in order to push down for input 1.

I understand that the internal wiring might look confusing, but the main take-away is that the "double bar" switches must be at the same height (y position) to produce the XNOR logic.

I didn't see a way how this can be achieved with simpler internal wiring. Do you have an idea?

richardabendroth, 2023-07-27:
I see, you are using a relay analogy. I was more thinking of a combination of serial/parallel arrangement of transistors where input and output negations basically just change the type of transistor (n/p): on when powered or off when powered. To be very honest, for me, it's harder to wrap my head around the relay analogy. It implies somehow a negation in one's head when thinking about it. The question is if the mnemonics should explain how an equivalent relay arrangement would work or what arrangement of transistors builds up the respective gate.