flandreas/antares #601: Discussion: Should gates support output negation as a property?

richardabendroth, 2023-06-28:
From my perspective, it would be nice for the learner if not only inputs could be negated but also the output of a gate, effectively allowing the user to turn an AND gate into a NAND gate and so on. What do you think?

flandreas, 2023-06-29:
One argument against this idea is that this would require a new property "Negate Output" in the properties window. Doesn't this give the learner the impression that an AND gate with a negated output is still an AND gate, while everybody else would then call it a NAND gate.

Sure, the title of the properties window would instantly change to NAND, but still: An AND gate with inverted inputs is still an AND gate, where as I think an AND gate with an inverted output is not an AND gate any more, hence probably not an operation to be backed-up by a property. I think I would rather prefer your suggestion #602 to change the type of a gate.

flandreas, 2023-08-10:
@richardabendroth Ok, I see, this leads to a similar discussion like in #598. When you look at logic gates, you clearly seem to have transistors in mind. Interesting..

richardabendroth, 2023-08-11:
I wouldn't necessarily say I have transistors in mind. More like electronic switches that are either make or brake and are arranged in certain patterns, like: 
- unary (nor BUFFER and NOT) -> number of inputs = number of outputs, one power source, n independent "switches"
- serial (for AND and NOR) -> always one output, one power source, "switches" in series
- parallel (OR, NAND) -> always one output, one power source, "switches" in parallel
- a combination of serial and parallel (XOR, XNOR) 
If you for example extend a serial or parallel gate to have more than two inputs, additional "switches" are added in between VCC and output, either in parallel or in series.  This could also be reflected in the mnemonics, just adding more "switches" connected to the respective output. When inverting one input for example this could also be reflected by changing the respective "switch" from e.g. make to brake. For reflecting an output negation the mnemonic changes basically from serial to parallel or vice versa. 
Does this somehow make sense?