flandreas/antares #606: Real Switch misbehaves on simulation startup

richardabendroth, 2023-07-21:
When starting the simulation of a circuit containing a "Real Switch", the switch shows as "on" but doesn't conduct. 

[image]

flandreas, 2023-07-21:
@richardabendroth I can't reproduce that.

https://github.com/flandreas/antares/assets/24633141/541f6fa9-71a3-4f1f-843d-c6a4494859a1

But I was surprised that it is possible that the switch is in state "on" right after simulation start. I guess that's a bug: It seems that it re-uses the last state from the previous simulation run.

However, even if the LED wire is pulled to 0 at startup, a signal conflict error occurs on that wire once to press the push button after start-up. Occurs only right after start-up, not later into the simulation when toggling both switches. Seems to be another bug.

May I ask you to send me this circuit for analysis? Maybe there are differences in propagation delays or something...

richardabendroth, 2023-07-22:
[PUPD.zip](https://github.com/flandreas/antares/files/12137089/PUPD.zip)

flandreas, 2023-08-11:
Fix: Make sure that switch is off at start of next simulation run

closed 2023-08-20 by flandreas