flandreas/antares #608: Register n Bit not working

XBlayz, 2023-08-06:
The component "Register n Bit" doesn't work, the output is always floating.

[image]

Version 1.5.0, Windows 11.

flandreas, 2023-08-08:
@XBlayz Thank you for reporting this.

The cause of the bug is that in the library component "Register n Bit", property "Always use function" is not set. It should be set, because it is a purely scripted component (no inner circuit).

Therefore, the component does nothing if the simulation is run with mode "View -> Deep Simulation". As a workaround, run your circuit without mode "View -> Deep Simulation" enabled, i.e. in mode "Shallow Simulation".

Will be fixed with the next release, upcoming soon.

XBlayz, 2023-08-08:
Ok, thanks for the help.

closed 2023-08-20 by flandreas