flandreas/antares #630: Multiplexer Issues

mrk67422, 2023-09-12:
A 2:1 2-bit multiplexer displays the switch graphics incorrectly when no bit is set to 1 at the input.

[image]

The 4:1 multiplexer does not transmit input states correctly. In practice, I see that the data inputs I0, I1 are duplicated. So if we choose I2, I3, it will show as for I0, I1

[image]

Similarly to the above with the 8:1 mutiplexer

[image]

By the way, I will also suggest using larger multiplexers and demultiplexers up to a level of e.g. 16:1 and 1:16

flandreas, 2023-09-12:
@mrk67422 Thank you very much for reporting this.

I'm not able to reproduce the problem with the 2:1 2-bit multiplexer. In the screenshot, I looks like a repainting problem. Does the problem go away when you zoom in or out, leading to redrawing the entire view?

I CAN however reproduce the problems with the 4:1 and 8:1 multiplexer. They only occur in "shallow simulation" mode, because the bug is in the simulation script of the components. Another occurrence of the nasty "type cast" pitfall. We'll fix this with the next release.

For your suggestion for larger plexers, I'll create a separate feature issue.

flandreas, 2023-09-12:
@mrk67422 Ok, I've tried the 2:1 multiplexer once more, you are right, and it's indeed a redraw problem. As you said, it occurs when changing S from 1 to 0 if input A is 0. E.g. changing the zoom factor corrects the wrong rendering. We'll try to fix that as well.

flandreas, 2023-09-12:
Repaint issue 2:1 multiplexer is too big to be fixed in a patch release. Created separate bug issue #632, targeted for release 1.7.

closed 2023-09-30 by flandreas