flandreas/antares #677: VHDL recommendation.

wmurbina, 2023-11-24:
When I try to export to VHDL a circuit the entity should have the same name as the circuit.

For example:

instead of main it should be Half_adder.

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

-- Half_adder
entity **main** is
  port (
    A: in std_logic;
    B: in std_logic;
    S: out std_logic;
    C: out std_logic);
end main;

architecture Behavioral of **main** is
begin
  S <= (A XOR B);
  C <= (A AND B);
end Behavioral;

flandreas, 2023-11-24:
@wmurbina Can you please explain why this is an issue for you? Does "main" conflict with something else, or what is your use case?

wmurbina, 2023-11-24:
The problem is when the generated code is to be used in other software such as Xilinx's Vivado.

It will be difficult to identify when you have several files.
[image]

Another thing that I have identified that is confusing, are the names given to the subcircuits when exporting in VHDL, they are too long, this could be improved by giving the possibility to put the name of the subcircuit (or a custom label) that is being used and that the entities take this name.

I attach an example of what I am trying to explain. 
[image]

flandreas, 2023-11-24:
Ok, I see, thank you!

The current naming scheme based on "main" and UUIDs for the subcircuits was straight-forward, and it ensured unique names. But I see your point.

So I will change it so that the entities are named like the circuits, adjusting them as needed to comply with VHDL naming syntax rules and uniqueness.

closed 2024-01-03 by flandreas