flandreas/antares #688: Error while loading rating data

Hougaarden, 2024-02-06:
I got this message when I tried to leave a rating (Windows 11)

[image]
.

flandreas, 2024-02-06:
@Hougaarden Thank you for reporting this. Yes we know, the backend cloud service we use for rating and usage counting are down since two days. We are currently looking into the issue; maybe we've exceeded some free usage quotas of the cloud service.

But thanks for trying to rate Antares :-)

flandreas, 2024-02-07:
@Hougaarden Backend services are back online, rating does work again.

closed 2024-02-07 by flandreas

Hougaarden, 2024-02-08:
Andreas,

Thank you for the quick response. The rating has already been given.
Too bad you can only enter 20 characters of text.
I would have liked to note that it is a pity that users cannot share their projects through the program, as in EveryCircuit, for example..

Regards, Wilfried