flandreas/antares #697: Problem with delay gates

fpw, 2024-03-29:
Hi,

I'm trying to simulate a rather complex digital circuit (~1000 gates spread across several sub-circuits) and when I did a little survey across the available tools, Antares seemed to be the best solution, mainly because it has great support for sub-circuits and tunnels. So thanks for this great tool!

But while debugging parts of the circuit, I found a problem with delay gates. The circuit I'm simulating is from the 1960s, so it uses delayed pulses instead of a clock to control gates. That's why there are many delay lines in the circuit.

The problematic part of the circuit feeds a short high-pulse into two parallel delay gates with 150 and 200 ns respectively. However, the 200 ns gate never forwards its input to the output. A single-step simulation revealed that the simulation is stuck waiting for the propagation delay while simultaneously waiting for the buffer reset of the faster line. 

I created a short video that demonstrates the issue, the two delay gates are near the center, I'm also circling around them with the cursor right before the issue is visible: The lower delay gate never stops flashing and never forwards the high input to its output.

I hope the video is useful to you to understand the problem. I could also provide the circuit, but it's not that easy to reproduce due to the number of  sub-circuits involved that need configuration.

https://github.com/flandreas/antares/assets/5798899/b053ec95-f436-42b9-8836-a0f0223ac780

fpw, 2024-03-29:
Thanks for taking the time to reproduce this! The subcircuit that produces INHIBIT is not supposed to have a zero propagation delay, it's basically just a different symbol for a flipflop that matches the original schematics I'm trying to simulate. It seems that the flipflop does have a propagation delay property of zero by default, but changing it always sets it back to zero. But the internal gates of the default flipflop seem to have delays anyways.

Here's the circuit and a clip that shows how to reproduce the issue inside the circuit:
* enable single stepping
* set simulation speed to full
* start simulation
* ignore the first breakpoint
* hit the "Cont" key, that will trigger the breakpoint again
* while paused, zoom into the MC component on the lower left
* set simulation speed to 25% and resume
* notice how the simulation never continues with the lower delay line flashing forever

[5ac5b066-4a77-4542-814d-d7bf19a8bbb6.zip](https://github.com/flandreas/antares/files/14802248/5ac5b066-4a77-4542-814d-d7bf19a8bbb6.zip)

[Clip that shows how to reproduce this](https://pdp8.app/antares2.mp4)

fpw, 2024-03-29:
I investigated this further and I think the speed change is causing the problem. Placing another breakpoint behind the delay gate in question shows that it's actually triggered if I run this at a higher speed. 
And I found several more instances where changing the speed mid-simulation makes everything stop, can provide more examples if needed.

flandreas, 2024-03-29:
@fpw Thanks for hinting towards speed change, I'll investigate in that direction...

fpw, 2024-03-29:
Thanks!
Meanwhile, I created a minimal example that reproduces the issue, completely independent of my circuit: 

https://github.com/flandreas/antares/assets/5798899/70f609c0-eb4d-4b4a-bf49-7e7ac1014145

Just dragging the speed slider randomly eventually stops the entire simulation. Even waiting for several minutes after the final frame of the video didn't make the lamp flash again.

flandreas, 2024-03-29:
@fpw Cool, you are a great investigator, that helps a lot, thanks!

flandreas, 2024-11-09:
Can also be reproduced with this even simpler circuit:
[image]

closed 2024-11-15 by flandreas