flandreas/antares #698: Suggestion: Working with tunnels in large circuits

fpw, 2024-03-29:
Hi,

Antares is capable of simulating huge circuits, that makes it really useful to analyze real-world circuits. The circuit I'm simulating is from the 1960s and uses wire wrapping, i.e. every part in the whole system can be connected to any other point without the need to create a connection on some kind of interface connector.

Speaking in terms of Antares, that means there are connections from a signal inside one sub-circuit to a signal within another sub-circuit without going through a port and a top component that connects them. It would be great if Antares supported that by using tunnels. Right now, tunnels seem to live in their own per-circuit namespace. And that makes sense because a sub-circuit can be instantiated multiple times.

But here are two different suggestions that could make it work:

## Singleton Circuits
* add an optional "singleton" parameter for a sub-circuit: this only allows creating a single instance of this sub-circuit. This also solves the problem that you can't change edit the sub-circuit components if you opened an instance from a top-circuit: There might be multiple instances, so Antares opens sub-circuits read-only when opening an instantiation. That's a little annoying to work with while debugging; if an error is found deep in a sub-circuit, you can't fix it in place but have to re-open the base circuit since the instance is read-only. For singleton circuits, this would no longer be a problem.
* add a "global singleton namespace" property to tunnels. If that's enabled on a tunnel that lives inside a singleton circuit, the same tunnel name can be used in other singleton circuits to create a direct connection without going through ports. 
* allow searching for tunnel names across all circuits in the project. In my thought model, tunnels are basically signal names and it's often useful to find where a signal is asserted and where it's used. A table that lists tunnel names with direction (in, out, undefined) would be great, then I could double click on a row with "in" type to see where the signal is asserted, etc.

## Regions in Circuits
A totally different approach that might be simpler, not sure:
* allow creating "regions" inside a circuit. Dragging a rectangle around a few components allows moving them into a region and will replace the components with a symbol for a region. Regions don't have connectors, they can use tunnels to communicate with the outside. Since a region lives in one and only one circuit, the tunnels inside them just share the circuit's namespace as usual.
* using regions instead of sub-circuits would be an alternate way to form a hierarchy: a sub-circuit is something that is intended to be reused, e.g. a flipflop, counter, etc. while a region is just a part of a bigger circuit, e.g. the ALU of a CPU. 
* still, the main point is a way to simplify parts of a bigger circuit connecting to a completely different part of the circuit through a tunnel without having to create a connection and port in a top component. The region approach wouldn't even need a top component, an entire CPU would be a single circuit with multiple regions that talk through tunnels.

## If that's not possible
In #697, you can see how I work around this by creating ports for every tunnel inside the sub-circuits. But that's quite a tedious work because I have to create a tunnel for every port to get the signal back into a local tunnel. If the suggestions are not possible, it would already help a lot if connecting a tunnel to a port would automatically copy the name from the 
tunnel to the port and vice-versa, maybe if a modifier key is held or something like that.

## Example Usecase
Here's the original circuit for reference: http://bitsavers.informatik.uni-stuttgart.de/pdf//dec/pdp8/pdp8i/PDP-8I_RevCY_Engineering_Drawings_Dec71.pdf

Page 14 is a good example, you can see that many gates simply end in tunnels - and these tunnels are used all over the place even on totally different pages. This is how the tunnel connections were implemented in reality: https://www.pdp8online.com/dfds32/pics/large/df32wirewrap.jpg

fpw, 2024-03-29:
For reference, I checked this in Multisim and it supports both:

Tunnels are called connectors there and they differentiate between on-page connectors (tunnels in Antares) and global connectors (what I'm missing). [Source](https://forums.ni.com/t5/Example-Code/Using-the-New-Multisim-11-Connectors/ta-p/4183371)

What I called region is actually called sub-circuit in Multisim, i.e. there is only one instance of a sub-circuit and it's more like a way to make a big circuit more readable. What Antares called sub-circuit is called hierarchical block in Multisim: Components that are stored in separate files and can be instantiated many times. [Source](https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z000000P9FpSAK)

Note that global connectors can't be used inside of hierarchical blocks, that way it makes sense to combine these features.

flandreas, 2024-03-30:
@fpw Thank you for this interesting suggestion.

First of all, one of the advantages of singleton circuits you've mentioned is editing sub-circuits after descending down the hierarchy. I'm afraid this is not possible with the current SDI architecture (single-document interface): The application can at any time only have one editable circuit, which is the top-level or main circuit. Changing this would require to change the architecture from SDI to MDI, which would be a challenge of its own.

However, I understand your use case. Personally, I like the approach with global connectors better than the "Regions in circuits". We could let the user define a global list of signals (on the project level), and when adding tunnel components to circuits, the user could link it with a global signal. Unlike Multisim, we would support this in sub-circuit as well, not only in the top-level circuit.

fpw, 2024-03-30:
That would be great. But doesn't the support for global signals in sub-circuits lead to conflicts? If the circuit is instantiated multiple times, it's possible that two instances write different values to a global signal. But thinking about this, it's already possible using normal tunnels by just creating two tunnels with the same name and writing different values to them, so I guess this is already handled like a normal write-write conflict - cool!

In that case, allowing a tunnel to become global seems to be a great and simple solution. Then my circuits wouldn't have external connectors at all and communicate with each-other through tunnels, as in the original sheets.

Speaking of tunnels, a minor thing which I noticed when parsing the circuit files myself to create a quick tool that lists incoming and outgoing tunnels of each circuit in a library: It seems the tunnel direction is only a property of the view and not of the model. Wouldn't it make sense to make it a property of the model? That way, the simulation could check violations, i.e. creating a conflict if an in-tunnel is "written" to.

flandreas, 2024-03-30:
@fpw On the model level, tunnels are essentially bi-directional. So, if the user knows what he's doing, he can indicate the primary (or only?) signal flow direction on the view level, but this is only for making the schematic clearer to the reader. The model layer will still allow signals to flow in both directions.

flandreas, 2024-04-04:
@fpw First implementation of global tunnels.

https://github.com/flandreas/antares/assets/24633141/ffc612de-5b70-4239-97a9-3d3c4affd9c5

fpw, 2024-04-04:
Looks awesome, thanks a lot! Looking forward to testing this 😃

closed 2024-04-05 by flandreas