flandreas/antares #707: Pasting a circuit with oscilloscope probes leads to null pointer exception

fpw, 2024-04-05:
After pasting a copied circuit that contained oscilloscope probes to a new circuit, starting the simulation caused the following error. Removing the oscilloscope probe fixed it.

Version: 1.10.0
Preparing paste from clipboard
Paste 364 components
Open application data
Copy 421 components
Open application data
Delete 359 components
Preparing paste from clipboard
Paste 421 components
Open application data
Open application data
Open application data
Enter execution mode (deep = true)
Enter edit mode
Enter execution mode (deep = true)
Enter edit mode
Open application data
Enter execution mode (deep = true)
Enter edit mode
Open application data
Enter execution mode (deep = true)
java.lang.NullPointerException
	at ch.scorpion.jabbah.graph.model.graph.GraphImpl$b.a(SourceFile:411)
	at ch.scorpion.jabbah.graph.model.a.a.a(SourceFile:100)
	at ch.scorpion.jabbah.graph.model.a.a.a(SourceFile:97)
	at ch.scorpion.jabbah.graph.model.d.d.a(SourceFile:51)
	at ch.scorpion.jabbah.graph.model.port.PortImpl.setIncomingSignal(SourceFile:156)
	at ch.scorpion.jabbah.graph.model.c.d.c(SourceFile:159)
	at ch.scorpion.jabbah.c.e.m.a(SourceFile:298)
	at ch.scorpion.jabbah.c.a.i.a(SourceFile:48)
	at ch.scorpion.jabbah.c.a.d.a(SourceFile:167)
	at ch.scorpion.jabbah.graph.view.u.a(SourceFile:3162)
	at ch.scorpion.jabbah.graph.view.o$a.b(SourceFile:112)
	at ch.scorpion.jabbah.graph.view.o.b(SourceFile:57)
	at ch.scorpion.jabbah.c.a.i.a(SourceFile:36)
	at ch.scorpion.jabbah.c.a.d.b(SourceFile:163)
	at ch.scorpion.jabbah.c.e.m$b.e(SourceFile:652)
	at ch.scorpion.jabbah.c.e.m.h(SourceFile:536)
	at ch.scorpion.jabbah.c.e.m.n(SourceFile:226)
	at ch.scorpion.jabbah.c.e.C.a(SourceFile:99)
	at ch.scorpion.jabbah.c.e.C.a(SourceFile:91)
	at ch.scorpion.jabbah.c.e.E.a(SourceFile:51)
	at ch.scorpion.jabbah.c.e.E.invoke(SourceFile:51)
	at ch.scorpion.jabbah.base.d.d.actionPerformed(SourceFile:14)
	at ch.scorpion.jabbah.base.p.c$a.actionPerformed(SourceFile:54)
	at java.desktop/javax.swing.Timer.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.Timer$DoPostEvent.run(Unknown Source)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

flandreas, 2024-04-05:
@fpw Thank you for reporting this, I've never tried to do this.

I haven't analyzed the issue yet, but when I'm thinking about requirements, I would say that oscilloscope probes should probably be excluded from paste operations. They kind of belong to the "configuration" of the origin circuit, and when pasted to another circuit, in the most general case, they could clash anyway with already existing probes in the destination circuit.

Would you agree?

flandreas, 2024-04-05:
Declared oscilloscope and its probe components as non-copyable

fpw, 2024-04-05:
Sounds like a good solution to me. I didn't copy them on purpose: My workflow was designing a sub-circuit and I used a scope to debug it. Then I copied the entire sub-circuit into my global circuit since there are no global tunnels yet, my current workaround is to just paste all sub-circuits into one big circuit. And since that doesn't have a scope, the issue appeared.

closed 2024-04-05 by flandreas