flandreas/antares #712: Can't edit custom symbol pins anymore in 0.11.0

fpw, 2024-04-15:
After updating Antares to 0.11.0, I can't edit the pin properties in my custom symbols: The properties window is empty and stdout shows this:

`2024-04-15 20:23:28,306  WARN [AWT-EventQueue-0] (AbstractPropertyPanelSwing.kt:197) - Could not instantiate Properties for ch.scorpion.antares.view.container.DigitalPortViewComponent: Exception java.lang.NoSuchMethodException: Unknown property 'largeExternalPortLabelDist' on class 'class ch.scorpion.antares.view.container.DigitalPortViewComponent'`

Screenshot shows how the properties view is empty:

[image]

flandreas, 2024-04-18:
Reproducible, thanks for reporting!

flandreas, 2024-04-18:
Fixed. This is a blocker, justifies a bugfix release soon.

closed 2024-04-19 by flandreas