flandreas/antares #714: Undo broken after opening RAM

fpw, 2024-04-16:
* Start Antares fresh so that it has a clean Undo stack
* Create a new circuit
* Add an AND gate
* Add a RAM block and double click it to see the contents, then close that window
* Delete the AND gate
* Undo
* Antares will show an unexpected error (dump below)
* But there's more: Close the circuit and working in a different circuit now always shows this error when doing Undo, the Undo stack is somehow poisoned now.

This might be related to #273 - I have a feeling that when this error is shown, Antares reverts the state back to some previous state. And that state seems to be asynchronously recorded because the number of gates that are still there depend on my placing speed or some other parameter I didn't understand - sometimes, both the RAM and the AND gate disappear and sometimes one of them is still there. If the state that Antares reverts back to is really recorded asynchronously, maybe it's possible that there's a race condition where the state is recorded between model and view changes, e.g. if a wire is placed and the view is updated before the model, is it possible that the state was saved _during_ this so that the recorded state doesn't contain the model connection, even though the view connection is already there? In that case, reverting to this state would cause #273.

Another possibility I thought of - just thinking aloud in the hope to trigger an inspiration for a fix - Since the undo stack is now in a situation where some action from a previous circuit is also present in a new circuit, is it possible that the undo stack is not properly cleaned between circuits? Maybe it's possible that something like this happens: Delete connection between vertex 2 and 3 in one circuit, hit undo in a new circuit and suddenly the connection between vertex 2 and 3 in the new circuit disappears. This doesn't happen under normal circumstances because the Undo stack seems to be cleared, but the fact that the error described in this circuit isn't cleaned up when switching circuits, maybe there's a possibility for something like this to happen.

Error dump:
```
Version: 1.11.0
Using workspace C:\Users\willfolk\AppData\Roaming\Antares
Close SideBarPane 'null'
Open SideBarPane 'Explorer'
Open project '5ac5b066-4a77-4542-814d-d7bf19a8bbb6'
Open main Library/Project 'PDP-8/I'
Open application data
Open application data
Add Component 'AND' from library/project at Point2D(-406.0,35.0)
Add Component 'RAM' from library/project at Point2D(-259.0,-161.0)
Delete component with ID 1
Undo command 'Delete'
java.lang.NullPointerException
	at ch.scorpion.antares.view.addressable.z.d(SourceFile:71)
	at ch.scorpion.antares.view.addressable.z.a(SourceFile:18)
	at ch.scorpion.antares.view.addressable.A.a(SourceFile:27)
	at ch.scorpion.antares.view.addressable.A.invoke(SourceFile:27)
	at ch.scorpion.jabbah.base.c.a.b(SourceFile:83)
	at ch.scorpion.jabbah.app.q.a(SourceFile:34)
	at ch.scorpion.jabbah.app.w.a(SourceFile:105)
	at ch.scorpion.jabbah.graph.h.w.a(SourceFile:88)
	at ch.scorpion.jabbah.edit.c.c.a(SourceFile:1066)
	at ch.scorpion.jabbah.edit.c.d.j(SourceFile:195)
	at ch.scorpion.jabbah.edit.a.N.a(SourceFile:40)
	at ch.scorpion.jabbah.base.j.actionPerformed(SourceFile:74)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Actions.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.SwingUtilities.notifyAction(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBinding(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(Unknown Source)
	at java.desktop/javax.swing.JMenuBar.processKeyBinding(Unknown Source)
	at java.desktop/javax.swing.KeyboardManager.fireBinding(Unknown Source)
	at java.desktop/javax.swing.KeyboardManager.fireKeyboardAction(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBindingsForAllComponents(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBindings(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
````

fpw, 2024-04-16:
Thanks a lot for the in-depth explanation! With that in mind, I will do a few more experiments in the hope to find a reproducible way to cause the wiring error.

flandreas, 2024-04-16:
BTW: If you are working on the development system, you could enable GraphViewConsistencyCheck in GraphModuleJVM.initialize() by removing the if() statement. This will check for the symptom in #713 with every executed Command, highlight the location of the issue in the circuit (if I remember correctly), and popup a Dialog.

closed 2024-04-19 by flandreas