flandreas/antares #724: Switch with internal label sometimes too narrow on Windows

flandreas, 2024-04-20:
Especially with short texts. See "IF0" etc below.

[image]

fpw, 2024-04-20:
As an alternate solution, it would be cool if the user could just drag the switch rectangle size themselves. For situations like the above, it would be more aesthetically pleasing if the switches all had the same size.

fpw, 2024-04-20:
My Java days are long gone, but I noticed that `KEY_FRACTIONALMETRICS` is not explicitly set on the `Graphics2D` context. Doesn't that mean that the implementation will decide what's best, i.e. depending on the DPI setting? 

This could be relevant because `TextRenderInfoFactoryJvm` just assumes that it's off when creating the `FontRenderContext` for measuring. For testing, I changed the last parameter from false to true and that's enough to make it look correct:

[image]

I believe the `FontRenderContext` should take the actual settings into account rather than making static assumptions. But as I said - not an expert on this topic, just a wild guess.

flandreas, 2024-04-23:
I can confirm that changing the parameter in `TextRenderInfoFactoryJvm` improves things on Windows, and macOS is still the same (correct). However, I've also noticed that on Windows, the overline's lengths depend heavily on the zoom factor. But this seems to be the case independent on the parameter in `TextRenderInfoFactoryJvm`.

https://github.com/flandreas/antares/assets/24633141/d4883cf8-4808-4834-aaf6-af5dd5b8c210

It also looks like font rendering by itself on Windows is quite unstable with changing zoom factors - with small zoom factors, text rendering gets quite "rough". Much rougher than on macOS.

I'm not an expert in these topics neither, so I guess we can change the parameter to improve some things, and others at least don't get worse.

fpw, 2024-04-24:
The rough zooming improves by actually enabling the rendering hint for fractional rendering. I suggest this:
````g.setRenderingHint(RenderingHints.KEY_FRACTIONALMETRICS, RenderingHints.VALUE_FRACTIONALMETRICS_ON)````
in combination with the `true` parameter in `TextRenderInfoFactoryJvm`.

closed 2024-05-05 by flandreas