flandreas/antares #726: Joining wires can lead to corrupt pin connections

flandreas, 2024-04-23:
This can be caused by joining open-ended wires or deleting wire junctions. Leads to pins not properly connected to the net of the wire, which results in simulation not working as expected.

Detected in #713.

flandreas, 2024-04-25:
Thanks! Reopened

flandreas, 2024-04-25:
Cannot be reproduced generally with a simple circuit having the same structure like the problematic region like in the video. 

However, it AN be reproduced using the dump provided in #713 with exactly the wire segment you've deleted. I'll investigate..

flandreas, 2024-04-25:
Reproducible when deleting the selected wire in this simplified circuit:

[image]

The problem occurs when the wire that gets joined is connected with yet another junction node.

fpw, 2024-04-25:
Great find, I think all instances in which I've encountered this bug previously were related to long wire junction ladders like this.

flandreas, 2024-04-25:
Unfortunately I think I've introduced this one with my last fix. The scenario above isn't reproducible in the productive version 1.11.

closed 2024-05-05 by flandreas