flandreas/antares #727: Focusable components should overwrite global hotkey actions

flandreas, 2024-04-25:
Extracted from #725 

Components like keyboard are currently not able to consume keys like space, because they are intercepted and consumed by top-level keyboard bindings. For example, the space key is used to pause and resume the simulation, preventing the user to enter spaces in the keyboard component during simulation.

fpw, 2024-04-28:
Great to see this fixed, thanks! 

There seems to be a small problem with it where sometimes hotkeys don't work anymore after stopping the simulation. Example:

- create a circuit with a keyboard
- start the simulation by clicking play
- focus the keyboard
- stop the simulation by clicking play

-> Now hotkeys like pressing delete to delete a component no longer work.

flandreas, 2024-04-28:
OK, thanks! I've forgot to clear the focus after the simulation has ended. Although I don't know yet how to solve this, because there can many independent simulations be running at the same time (using "Open in new window.."), but there is only one global focus manager. I have to think about that..

flandreas, 2024-05-01:
@fpw Fixed by resetting the FocusManager after simulation has stopped. This leaves the issue that a second open simulation window looses its focused component, but I think we can live with this corner case for now.

closed 2024-05-05 by flandreas