flandreas/antares #730: Unclocked RAM write is not edge-triggered

fpw, 2024-04-30:
While debugging the last RAM issues in my circuit, I noticed another oddity: If a RAM component is unclocked, the write signal is level-triggered instead of edge triggered. That is, if I assert the write signal and leave it asserted, every change of the data or address will cause additional writes with to the new address / with the new data. 

I don't think this is intentional, the write signal should be edge-triggered instead, i.e. ignore further changes to the inputs while write is asserted.

If there are no objections, I will file a PR to change the behavior.

flandreas, 2024-05-01:
This behaviour is indeed intentional. In this asynchronous mode, the RAM behaves like an array of latches, where the address input is directly connected to the decoders, and the data input is directly connected to the D input of the latch.

This is also like RAM (with asynchronous interface) works in Logisim and the asynchronous RAM works in Hneemann Digital.

Why don't you use the synchronous RAM?

fpw, 2024-05-01:
I understand that compatibility across these tools is important, so I'll just add a pulse generator to my write externally.

closed 2024-05-01 by fpw