flandreas/antares #731: Image switch component

flandreas, 2024-05-03:
A version of a (digital) switch component having two images, one for each of its "on" and "off" state.

Based on the image capability provided with #729.

flandreas, 2024-05-03:
Preview of current implementation using SVG images. Making progress..

https://github.com/flandreas/antares/assets/24633141/9ded004f-5afd-42d4-ad29-0121d5b65317

fpw, 2024-05-03:
Wow, great to see that you chose to support SVGs! One of the things I like about Antares is how everything is vector based and I can zoom as far in or out as I want. Including custom SVG art will allow amazing designs.

On a side note, I'm also working on switches right now - but it's mainly in the model so there shouldn't be conflicts.

closed 2024-05-05 by flandreas