flandreas/antares #739: No hourglass mouse pointer while opening large circuits

flandreas, 2024-05-06:
When double-clicking on circuit nodes in the explorer tree to open the circuit: If the circuit is large, there may be no system response for seconds, until an hourglass mouse pointer is displayed immediately before the circuit is displayed. The same happens when ALT-clicking on a subcircuit symbol to open it in a secondary view.

The hourglass mouse pointer should be displayed immediately after the user double-clicked.

fpw, 2024-05-06:
I noticed the slowdown in my PDP-8 circuit as well - but it was only introduced when I added more and more images. Before the image feature, opening was as fast as opening a small circuit. Maybe the parsing of the SVGs is somewhat slow and could need a kind of global cache from UUID to whatever object results from the parsing?

flandreas, 2024-05-07:
Thank you for pointing towards image loading.

Turned out it wasn't so much the SVG loading itself, but more the way the library tree was traversed when looking for the image.

- Fixed the missing hour glass pointer
- Improved the way the image is searched
- Introduced a cache

closed 2024-06-14 by flandreas