flandreas/antares #762: Error 404 when trying to download examples

logoliv, 2024-06-21:
When I click on the download buttons in the https://www.antarescircuit.io/docs/examples/examples/ pages I obtain a 404 error.

Another question : what's the login inside the application for ? And how to create an account ? What's the purpose of that ?

flandreas, 2024-06-22:
Thank you for your feedback!

Download links: Fixed.

Disable tooltips: Okay, that makes sense, I'll create a separate issue for this. There is already the "Show tips for beginners" user preference, but I didn't think of truth tables and component name tooltips also to be disabled.

Pan: I understand your request such that you'd like another "Pan method" option in the View user preferences for "space + left click".  I'll create  a separate issue for this. Just out of curiosity: Why do you think middle click is not practical? That's the option I'm personally using, also in other applications that support panning.

logoliv, 2024-06-22:
Thanks for your answer and for fixing download links :)

I think it's a question of personal preferences concerning the middle mouse button : I have no problem using the mouse wheel but I don't feel comfortable using it as a button. Some applications that don't have contextual menus (but Antares has) give the option to use the right button for pan and it feels OK through... But I certainly must not be the only one with the same feeeling.

I've also noticed that pressing the alt key highlights the file menu : you don't have menus shortcuts (and it's good like that) so you don't need that, maybe disable it if you find the instruction to do that, but it's just a detail.

And at last: thanks for developping Antares ! I was searching a replacement solution for the old Logisim, and your app has a good design, a nice feeling and already some nice functionalities :)

closed 2024-06-22 by logoliv