flandreas/antares #765: Custom text color lighter than expected when using Darcula theme

logoliv, 2024-06-22:
Here's an example when I choose "green" as color, but it's the same for all colors when using this theme :

[image]

flandreas, 2024-06-22:
@logoliv Thank you for pointing that out, I wasn't aware of this.

The reason why it currently behaves like that is that the text color has been hand-crafted for applications like rectangles with text, where there is a coloured background of the same color type. Obviously, this doesn't work very well with plain texts without background.

[image]

I'll think about what a good strategy for choosing the text color in this use case could be. Maybe the background color..

flandreas, 2024-06-22:
I'll go with the foreground (i.e. border) color:

[image]

logoliv, 2024-06-22:
nice 👍

closed 2024-06-22 by logoliv