flandreas/antares #769: Request : bit-extender with 1 bit output width

logoliv, 2024-06-25:
For my actual project I made a subcircuit that is identical to a D Flip-Flop PRE/CLR (neg) but with an enable pin and parametrable bit width. It works fine but I can't use it with a 1 bit BW (the selection just doesn't work).
I think the cause of the problem is the bit-extender component that don't accept 1 bit output (and that's logical for a bit extender) :

[image]

It would be helpful to make this component also accept 1 bit output : it's mandatory if we want to make subcircuits that can accept a BW from 1 to 32 bits. Here is my circuit if you want to test by yourself.

[D Flip-Flop with Enable.zip](https://github.com/user-attachments/files/15970235/D.Flip-Flop.with.Enable.zip)

flandreas, 2024-06-25:
You're right: The bit extender's current rule is that output bit width must be > than input bit width. Equal bit width is not accepted.

I don't see why equal size shouldn't be accepted. How urgent is this for you and your project (in terms of release date?)

logoliv, 2024-06-25:
ASAP ? No just a joke :)

I've no "release date" yet, it's a middle term project I have as a hobby : I'm recreating the Z80 CPU from schematics I found for Altera Quartus... My turn to ask : what's your estimated release date for 1.14 milestone ?

I could make a separate 1 bit version (if I want to get rid of the tri-state buffer that I add for now before the clock pin of the standard library DFF). Just proposed that so Antares can have more generic components, usable in more situations.

flandreas, 2024-06-26:
Perhaps end of July. I'm glad you have a workaround.

Interesting, your project with the Z80! Keep asking for missing features if you get stuck, and I would love to see your project someday. Maybe you present it in the discussions once it has reached a level you're happy with :)

flandreas, 2024-06-26:
@logoliv You're welcome.
BTW you don't have to close issues when I've set the "done" label. I usually keep the open until I deploy the release they are targeted for. This way people see what's coming with the next release, which also leads to fewer duplicates.

logoliv, 2024-06-26:
OK I was asking myself 😃

closed 2024-08-09 by flandreas