flandreas/antares #774: Suggestion : add default value for pins

logoliv, 2024-06-28:
I think it would be very useful to implement a defaut value for input and output pins.

For example in the schematics I try to reproduce in Antares there's a regular use of the "D Flip-Flop with enable" that I described in issue #769 but the PRE and CLR pins are generaly not wired, they are initialised by default to 1 (as they use negative logic).

With script it's easy to set their value in the init part, but the equivalent in circuit would be fine, and it would avoid me to do this everytime I use it (same for the enable pin) : 

[image]

flandreas, 2024-06-28:
Well, I know what you mean, but this topic is kind of controversial. There are other simulator tools that strictly refuse to start simulation if there are unconnected inputs. Other tool treat unconnected inputs with random input values, just as what you would experience with a real circuit when waving with your hand over the circuit.

Antares has a user preference "Circuit -> Digital -> Undefined gate input behaviour" with options "Read as 0", "Read as 1" and "Read as random", but this isn't yet applied to subcircuit pins.

I could use the user preference, apply it to subcircuit pins, and invert the signal if the pins logic is negative. This would work for your use case, but not when you open the subcircuit as main circuit: Then, you would probably control it separately, and not only for inputs, but also for switches.

I have to think about it.

logoliv, 2024-06-28:
Logisim-evolution for example does that for each pin individually, not globally in the preferences : 

[image]

Your solution is elegant and would be fine in my case for the PRE / CLR pins, but not for the enable pin that should have a default value 1 and has positive logic (the case where the E pin is not wired is less frequent through).

flandreas, 2024-06-29:
> ...but not for the enable pin that should have a default value 1 and has positive logic

Okay, I see, thank you.

closed 2024-08-09 by flandreas