flandreas/antares #780: Error message "Input port not found"

logoliv, 2024-07-02:
I just defined a 100% scripted subcircuit with an 8 bit input (inp) and a 16 bit output (out). The script is :
```
init {
out = 0
}

out@7 = inp@7
```
When running the simulation I have the issue "Input port 'out' not found at 0:0".
If I just replace the last line by "out = in" everything is fine.
I need some help here... 

Subcircuit joined [nbits output.zip](https://github.com/user-attachments/files/16066696/nbits.output.zip)

logoliv, 2024-07-02:
Same error message in this situation : data is input 16 bit, address is output 16 bit
```
var adr = (address and 0b0000000001111111)
address = (data and 0b1111111110000000) or adr
```
I need in a certain case to replace the 9 higher bits from address by the ones from data. No way to find a workaround (I first tried with a for loop and bit accessor with address@i = data@i, as you see I also tried with an intermediate variable).
It's the reference to an output that causes problem I think : if you only keep "address = (data and 0b1111111110000000)" it works.

You should prioritize the resolution of this bug for 1.14, as I have the impression it will quickly become blocking for my project...

flandreas, 2024-07-03:
The problem is that the bit accessor works on variables, but not yet on pins

logoliv, 2024-07-03:
There's no accessor in my second case, it's just an AND between numbers. Plus I've already tried something like : 
```
var adr = address
address = (data and 0b1111111110000000) or (adr and 0b0000000001111111)
```
and I doesn't work either, though I use a variable for the AND instead of the output pin.
The problem appears as soon as I use the output pin as reference to do some operation.

flandreas, 2024-07-03:
Yes, the problem is reading values from output pins. Scripting in Antares so far only supports reading values from input pins.

The bit accessor needs to read the output pin to get the value in which a bit has to be changed.

Your second scenario is not so clear. Since address is an output: What value would you address to possess the first time the script is executed? Any, my fix will at least avoid that the second scenario doesn't produce an error.

logoliv, 2024-07-03:
Here is the component stripped from all the circuitry : [Copy of inc dec.zip](https://github.com/user-attachments/files/16080810/Copy.of.inc.dec.zip)  
It's an incrementer/decrementer and the limit6 input limits the carry propagation to the bit 6, so that bits 7 to 15 are left untouched. I always try to make a script in addition to the circuitry of the medium complexity components, to optimize speed in case of "flat simulation".

logoliv, 2024-07-05:
It's fine when i work with an intermediate variable instead of the output : 
```
if ('carry in') {
	var adr = 0x0000
	if (decrement) adr = data - 1
	else adr = data + 1

	if (limit6) address = (data and 0b1111111110000000) or (adr and 0b0000000001111111)
	else address = adr
}
else address = data
```