flandreas/antares #785: Transistors represented as open switches during simulation

logoliv, 2024-07-06:
When choosing "inverter" as transistor symbol : during circuit creation symbol is fine, but during simulation it appears as open switch everytime (even when gate is closed) : 
[image]
We don't need to have a "switch", it's disturbing, and the change of color of the gate is sufficient to see that the transistor is open or closed. Keep the same symbol as in edit mode.

logoliv, 2024-07-08:
For the floating output, you can color blue what will be the "closed switch" from D to S, that would be fine as when a transistor's gate is open the high impedance state begins just after the source anyway.

flandreas, 2024-07-11:
That's a bug, thanks for reporting.

Regarding the question whether we need a switch or not: I personally like the switch, because it helps me to mentally track what's going on. And I guess the same is true for students.

But I understand that these features can be disturbing for pro's. Maybe this feature should be made optional. There are already several setting that turn certain "supportive" features on and off:
- Simulation -> Animation
- View -> Gate Mnemonic
- Preferences -> General -> Show tips for beginners

I'm not sure which of these settings to choose for enabling/disabling transistor switch display.

closed 2024-08-09 by flandreas