flandreas/antares #806: Issue while using 8-bit architecture modules

Krishanu-007, 2024-08-23:
Continuous popping of System Malfunction, whenever any 8-bit module is being used forcing the app to restart

Krishanu-007, 2024-08-23:
[Antares.log](https://github.com/user-attachments/files/16728271/Antares.log)
[currentStorable.txt](https://github.com/user-attachments/files/16728272/currentStorable.txt)

flandreas, 2024-08-23:
@Krishanu-007 Thanks for reporting.

You seem to have exported a system dump file. It should be named something like Antares-dump.zip. Could you please attach it to this issue? I'll need it for analysis.

Krishanu-007, 2024-08-23:
[Antares-dump.zip](https://github.com/user-attachments/files/16729683/Antares-dump.zip)

flandreas, 2024-08-23:
Thanks! I can reproduce the issue. I still have to find its cause, and then I'll release a fix as soon as possible..

flandreas, 2024-08-23:
@Krishanu-007 Seems to be a "false positive" of the consistency checker infrastructure. This has to be fixed.

In the meantime, I've disabled the consistence checker. Disabling is controlled by a file Antares reads from the GitHub repository. Make sure to restart Antares and try again. I've just tried it with your project (contained in the dump you've posted), and I was able to successfully add the "Multiplier n bits" component to your circuit "8-bit Arithmetics".

closed 2024-08-25 by flandreas