flandreas/antares #865: Suggestions for Enhancing Anteras with New Features

Abdullah-dev99, 2024-12-31:
Dear Flandreas:
I have been using Anteras for quite some time to explore and learn digital circuit design and computer architecture. I sincerely appreciate the functionality and user-friendliness of this software—it has been an invaluable tool in my journey.
I would like to propose a few feature enhancements that I believe could further enrich the user experience and make Anteras even more engaging and efficient for its users:

Power Rename Functionality:
Renaming inputs and outputs can sometimes be cumbersome when it requires manually locating and selecting them in the circuit. I suggest introducing a "Power Rename" feature—a new window in the Explorer that lists all existing input and output names, allowing users to rename them directly. Selecting an input or output in this window could highlight its corresponding location in the circuit for ease of identification.

Enhanced Number System Support for RAM/ROM:
Currently, addressing and data entry in RAM/ROM is limited to the hexadecimal number system. Adding the ability to edit and address RAM/ROM using multiple number systems (Binary, Octal, Decimal, Hexadecimal) would significantly improve usability and flexibility.

Persistent Storage Device:
Including a storage device component would be immensely beneficial for many projects. While RAM is volatile and its data is erased after simulation, and ROM retains data but is not editable during simulation, a storage device that is both editable during simulation and retains data after the simulation or software is closed would greatly enhance functionality.

Component Visibility in Symbol Mode:
In cases where a circuit is designed for a specific component (e.g., a Video RAM adapter), aligning inputs and outputs with respect to that component is critical. However, components are not currently visible in Symbol Mode, requiring repeated manual adjustments. Allowing components to be visible in Symbol Mode—strictly for the purpose of port placement—would make this process much more convenient and accurate.

flandreas, 2025-01-01:
Thank you for your feedback and your feature suggestions.

_Power rename functionality:_
Interesting. I've never saw the need for such a feature in my own design. Perhaps your circuits are larger than mine. I understand what you are asking for, but let me first clarify:
- Are you aware of the "Find" feature (Cmd-Ctrl-F), where searching for a input/output name select the search results in the circuit?
- Does your need arise from the fact that inputs/outputs have initially default names like "I5" or "O2", which you all have to change at some point? Support for adjusting the default name pattern was requested by another user in issue #612 .
But anyway, having an overview of all inputs/outputs would surely be useful, especially for large circuits, and editing the names there would definitely make sense.

_Enhanced number system support for RAM/ROM:_
 I assume you are talking about  a ROM/RAM's "Content" view. Would you want a drop-down of the "Content" view where the user can choose "Representation" with all the supported number system, which is then applied to address row headers as well as the table cells? Or would you also want to be able to choose the number system when entering data in a cell, e.g. by entering "0x23"?

_Persistent Storage Device_
RAM has a property "Non-volatile" that supports probably just what you are looking for.

_Component Visibility in Symbol Mode_
Okay, I see your requirement. That makes sense. I would have to think about what a good UI solution could look like. For example, have a copy of the explorer tree as a right-side pane in the symbol editor with an enlarged preview? Or do you have ideas how we could solve this?

Abdullah-dev99, 2025-01-02:
Dear Flandreas
Thank you for your detailed response and for considering my suggestions. I truly appreciate the time and thoughtfulness you’ve put into addressing each feature.

I also want to take a moment to commend you for creating Anteras. This is an incredible project, and I understand how challenging it must be to build and maintain such a powerful tool. Your work is making a real difference for enthusiasts and learners like me, and I’m deeply grateful for your dedication and innovation.

Regarding your feedback:

Persistent Storage Device: Thank you for pointing out the "Non-volatile" property of RAM! It seems to address my need for data persistence. I’ll explore it further.

Enhanced Number System Support for RAM/ROM: The ability to input data in various formats, alongside the "Content" view enhancements, would be very helpful. Adding a side panel or tab on the right—similar to the scenario/use-case side panels—could make it even more intuitive by displaying all ROM/RAM content and showing which byte actions are performed during simulation.

Component Visibility in Symbol Mode: While your proposed right-side pane with an explorer and real-time preview is a great idea, I think it might not fully address the issue. For precise alignment, if Component A’s inputs need to correspond to Component B’s outputs, having Component B as just a temporary preview in Symbol Mode would be more effective than simply viewing it in a tree pane. This approach could streamline the process and avoid redundant steps.

Once again, thank you for your hard work and for listening to user feedback so attentively. Anteras is an extraordinary tool, and your commitment to improving it is truly inspiring. I look forward to seeing how the software evolves and am always happy to share additional thoughts if needed

flandreas, 2025-01-03:
Thank you for your kind words, much appreciated.

I've created the following new separate issues that reflect your requests:
- Power rename functionality: #870 
- Component visibility in symbol mode: #869 
- Enhanced number system support for RAM/ROM: #868 

A remark to your comment in "Enhanced number system support":

> Adding a side panel or tab on the right—similar to the scenario/use-case side panels—could make it even more intuitive by displaying all ROM/RAM content and showing which byte actions are performed during simulation.

If you ALT-doubleclick on a RAM/ROM, a new desktop view showing the memory's contents is opened. This also works during simulation.

flandreas, 2025-01-03:
@Abdullah-dev99 Current UI design idea for "Component visibility in symbol mode": Display a ghosted, non selectable version of the symbol to which the current symbol is to be compared to inside the drawing area. The user can choose at which side the ghost is displayed (north, east, west, south).

When zooming, panning or resizing the view, the position of the ghost is adjusted such that it always sticks to the selected border of the view.

In the sidebar to the right, the explorer tree is still missing. Upon selection of an element in the explorer tree, the ghost would be exchanged.

[image]

closed 2025-02-09 by flandreas