flandreas/antares #880: consistent crash when adding a joystick

DS-AdamMilazzo, 2025-02-05:
I consistently get a crash when trying to add a joystick component. All I have to do is:
1. Open Antares to a new, empty circuit.
2. Try to add a Joystick from the standard library.

The error is:
```
Version: 1.16.0
Using workspace C:\Users\Adam.Milazzo\AppData\Roaming\Antares
Close SideBarPane 'null'
Open SideBarPane 'Explorer'
Open main Library/Project 'My first project'
Open application data
Add Component 'Joystick' from library/project at Point2D(-539.0,-70.0)
java.lang.IllegalArgumentException: rotation not supported
	at ch.scorpion.jabbah.edit.model.AbstractComponent.setRotation(SourceFile:145)
	at ch.scorpion.jabbah.graph.view.vertice.AbstractVerticeView.setRotation(SourceFile:305)
	at ch.scorpion.jabbah.graph.view.a.a.a(SourceFile:35)
	at ch.scorpion.jabbah.edit.c.d.a(SourceFile:256)
	at ch.scorpion.jabbah.edit.c.d.a(SourceFile:170)
	at ch.scorpion.jabbah.graph.view.a.d.a(SourceFile:144)
	at ch.scorpion.jabbah.graph.f.F.b(SourceFile:96)
	at ch.scorpion.jabbah.edit.j.a(SourceFile:1127)
	at ch.scorpion.jabbah.edit.k.a(SourceFile:90)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
```

flandreas, 2025-02-05:
@DS-AdamMilazzo Thank you for reporting this. Will be fixed with the next release which is scheduled soon.

DS-AdamMilazzo, 2025-02-05:
Thank you for the quick response.

closed 2025-02-09 by flandreas