flandreas/antares #882: Crash when calculating propagation delay in 1.16.0

DS-AdamMilazzo, 2025-02-09:
When I ask Antares to calculate the propagation delay, it sometimes crashes. Here is a consistent repro:

1. Create a new circuit.
2. Add two inputs and one output.
3. Add an AND gate and wire it to the inputs and output.
4. Calculate propagation delay.

```
Version: 1.16.0
Using workspace C:\Users\Adam.Milazzo\AppData\Roaming\Antares
Close SideBarPane 'null'
Open SideBarPane 'Explorer'
Open main Library/Project 'My first project'
Open application data
Open application data
Add Component 'Circuit Input' from library/project at Point2D(-441.0,-133.0)
Add Component 'Circuit Input' from library/project at Point2D(-441.0,-28.0)
Add Component 'Circuit Output' from library/project at Point2D(-175.0,-84.0)
Add Component 'AND' from library/project at Point2D(-252.0,-77.0)
Start creating new EdgeView at Port 1 of VerticeView 1
Connect output 1 of 1 with input 1 of 4
Start creating new EdgeView at Port 1 of VerticeView 2
Connect output 1 of 2 with input 2 of 4
Start creating new EdgeView at Port 3 of VerticeView 4
Connect output 3 of 4 with input 1 of 3
java.util.NoSuchElementException
	at ch.scorpion.jabbah.graph.model.graph.d.a(SourceFile:15)
	at ch.scorpion.jabbah.graph.f.e.a(SourceFile:22)
	at ch.scorpion.jabbah.base.j.actionPerformed(SourceFile:73)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
```

flandreas, 2025-02-09:
@DS-AdamMilazzo Thank you for reporting. Can reproduce with 1.17.0 too.

It seems the error only occurs as long as the circuit has not been saved yet. Once it has been saved, the function "Calculate propagation delay" produces the correct result. Still needs to be fixed though.

DEV note:
```
2025-02-09 20:30:41,606 ERROR [AWT-EventQueue-0] (InteractiveErrorHandler.kt:34) - Unexpected error: null
java.util.NoSuchElementException
	at ch.scorpion.jabbah.graph.model.graph.GraphPropagationDelayCalculator.calculate(GraphPropagationDelayCalculator.kt:15)
	at ch.scorpion.jabbah.graph.ui.CalculatePropagationDelayAction.execute(CalculatePropagationDelayAction.kt:22)
```

flandreas, 2025-02-10:
Fixed. Moved action from explorer node context menu to "File" menu.

closed 2025-02-28 by flandreas