flandreas/antares #883: Crash when trying to delete the folder for a newly created library

DS-AdamMilazzo, 2025-02-11:
I created a new library with no import, right-clicked on the library item under "Desktop", and chose "Delete folder". After confirming, it crashed.

```
Version: 1.16.0
Export '2:1 Multiplexer as bundle
Export '2:1 Multiplexer as bundle
Delete 7 components
Delete component with ID 8
Max snapshot size reached. Create new snapshot.
Rotate 1 components, clockwise = false
Rotate 1 components, clockwise = false
Rotate 1 components, clockwise = false
Move component 'Pin' with ID 9 by mouse
Move component 'Origin' with ID 6 by mouse
Descending into SubGraphVerticeView
Add Component 'Image switch' from library/project at Point2D(-651.0,140.0)
Delete component with ID 89
new library
creating new library 'Test'
Create new library 'Test' with import 'null'
Container (Symbol) has been customized manually
Open main Library/Project 'Test'
Delete folder 'Test'
Delete folder 'Test'
java.lang.IllegalStateException
	at ch.scorpion.jabbah.graph.library.aY.c(SourceFile:326)
	at ch.scorpion.jabbah.graph.library.aY.a(SourceFile:184)
	at ch.scorpion.jabbah.graph.library.I.a(SourceFile:44)
	at ch.scorpion.jabbah.base.j.actionPerformed(SourceFile:73)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
```

flandreas, 2025-02-11:
@DS-AdamMilazzo Thank you for reporting this. I think the action shouldn't be enabled on the library node.

flandreas, 2025-02-11:
Fixed by disabling the action on the library node.

closed 2025-02-28 by flandreas