flandreas/antares #945: Antares locks up on "Save"

gt6796c, 2025-06-23:
I've got a fairly new project with a couple of custom circuits that use some of the 74xx chips. I'm trying to make a circuit that uses two of my other circuits and one of the 74xx chips. For whatever reason, when I hit "Save" the app just locks up. The log file's last entry is just `2025-06-23 13:10:24,386  INFO [AWT-EventQueue-0] (SourceFile:234) - Save application data` No errors or messages are sent. This is on version 1.21 and macOS. It happens 100% of time on this circuit now.

flandreas, 2025-06-26:
@gt6796c Thank you for uploading your project.

The bug is in the logic that automatically calculates the propagation delay of the circuit. It iterates over all possible paths from inputs to outputs and takes the maximum. It seems that this does not yet work correctly with bi-directional pins, as it can lead to infinite loops. I'll have to fix that with the next release.

As a workaround, you can set property "Propagation Delay (ms)" in circuit "register", which has currently the value 260 (automatically calculated during a previous save), to another value, e.g. 300. By doing so, you explicitly define the propagation delay of the circuit, and Antares doesn't try to calculate it anymore when the circuit is saved.

flandreas, 2025-06-26:
I'm sorry for the inconvenience you're facing.

Yes, i also think it's not the primarily the SN74245, it's potentially every subcircuit that has bidirectional ports, potentially leading to infinitely enlarging the paths it examines to find the longest one.

Maybe you should hold on until this is fixed. Or, you could save often, and if the problem occurs, revert to the last stable version of the circuit using Explorer -> Context menu on circuit -> Show history...

flandreas, 2025-06-26:
.. or, when I think about it, it can really be any subcircuit you use in a way where one of its output pin is connected to one of its input pins.

flandreas, 2025-06-26:
I've deployed a test version containing a fix for this problem. I've successfully tested it on the project you've uploaded.

If you'd like to test it, install as follows:

- Go to https://github.com/flandreas/antares/releases/tag/v1.21.0
- Download antares-1.22.0-preview1.jar
- Rename it to antares-1.21.0.jar
- Overwrite the existing JAR file in your Antares application installation. On my Mac, this is /Applications/Antares.app/Contents/app. You may have to sudo depending on your settings.

I've also added a user preference in Circuit -> "Calculate propagation delay upon save" that you can use to disable the feature if the fix doesn't work properly.

gt6796c, 2025-06-26:
I've been in there playing around for about an hour in a few different circuits and no issues! About to get to the point of hooking up a couple of things and test the simulation.

gt6796c, 2025-06-27:
I spent a good two hours in it last night with no issues at all. I got my "main" circuit going with using the bidirectional pieces from sub circuits and it is all working well. I was able to simulate a couple of the registers in that project with an ALU circuit all communicating on the bus wiring up to the main circuit with manual inputs and the like. 

Thanks a lot for the project and fixes!

flandreas, 2025-06-28:
Great, I'm glad to hear that. This means I can soon make a regular release with the fix.

I'm also happy to hear that simulation of your heavy-bidirectional project seems to work so far. The simulation algorithms in the bidirectional area are not for the faint-hearted, and in case you run into inconsistent behaviour, please file an additional issue.

closed 2025-06-28 by flandreas